InfiniTimeOrg/InfiniTime #1783: Weather: temp won't display in Fahrenheit

5iver, 2023-06-26:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

Weather only reports in Celsius

### What should happen instead?

When specified, weather should report in Fahrenheit

### Reproduction steps

I am using Gadgetbridge (0.74.0) with the QuickWeather (2.3.2) app on Android. GadgetBridge is configured for Imperial and QuickWeather is configured for Fahrenheit. My family all has PineTimes running InfiniTime 1.13.0 and are displaying the temperature in Celsius.

### More details?

I do not see any relevant issues reported for either of their apps. A quick look through their repositories did not lead me to believe the issue was elsewhere, especially because PineTime recently implemented weather functionality. Though, I could absolutely be incorrect and apologize in advance if that is the case.

Has anyone seen Fahrenheit temperatures using the new weather feature?

### Version

1.13.0

### Companion app

Gadgetbridge 0.74.0

kieranc, 2023-06-28:
This is correct, there's no imperial support yet, I believe it should be fairly easy to implement so I'll leave some notes here in case anyone gets to it before me.

[This](https://codeberg.org/Freeyourgadget/Gadgetbridge/src/commit/cc9c8d4207f746d5df79f62ca7b90f2d624a30fe/app/src/main/java/nodomain/freeyourgadget/gadgetbridge/service/devices/pebble/PebbleProtocol.java#L1151) is how it's implemented for the Pebble. It checks if GB is set to use imperial units and if so, recalculates the temperature sent to the watch. The watch deliberately doesn't include units after the degrees symbol so it should display fahrenheit just fine, at least below 99 degrees, after that there probably isn't space to show 3 digits. So don't go anywhere too hot.

[This](https://codeberg.org/Freeyourgadget/Gadgetbridge/src/commit/cc9c8d4207f746d5df79f62ca7b90f2d624a30fe/app/src/main/java/nodomain/freeyourgadget/gadgetbridge/service/devices/pinetime/PineTimeJFSupport.java#L753) is the relevant section for Pinetime.

Brod8362, 2023-06-28:
@kieranc and I talked a bit on discord and I want to summarize our conversation here:

- It seems like it could be possible to approach this from either the companion app side, or performing the conversion on the watch
- By performing it on the watch, any companion app that publishes weather data should be supported
- It should be easy for the user to switch their preferred unit (e.g no recompiling)
- For now, it seems like just having another setting is ideal. 

I'm still familiarizing myself with the WeatherService and how it is used, but it should be easy enough to change `WeatherService::GetCurrentTemperature` to just convert to fahrenheit within that function (when set). The watchfaces wouldn't have to do anything special to support this (aside from being prepared to render 3 digits).

Currently, though, I'm looking into simulating the weather controller service from within InfiniSim, to make development for this easier.

FintasticMan, 2023-06-28:
I would personally prefer the conversion to be done on the watch, rather than on the companion app. I would say we should then add a `GetFahrenheit()` function to the `WeatherData::Temperature` class.

Brod8362, 2023-06-28:
That seems like a pretty elegant solution. I'm mostly working on trying to see how feasible it will be to add weather integration to InfiniSim.

PR #1805 (2023-12-23): PTS: Add setting to display temperature in Fahrenheit
This adds a "weather format" setting, which can be either Metric or Imperial.

Fixes #1783.

@kieranc does this look good to you?
closed 2023-12-23 by JF002