InfiniTimeOrg/InfiniTime #1784: Heart measurement result is zero on watch face after blank screen

ajira86, 2023-06-28:
### Verification

- [X] I searched for similar issues and found none was relevant.

### Introduce the issue

**How to reproduce**

* Open Heart app
* Start the measurement
* Move to watch face
* Wait for screen to suspend
* The heart measure displays zero after a few seconds (about 5 to 30)

It seams to be only affected by blank screen. If user change screen and come back to watch face, the value don't change to zero

### Preferred solution

The measure value should not display zero value in this case but should act as if we are switching app and come back to watch face screen.

### Version

v1.1.13

FintasticMan, 2023-06-28:
This is expected behaviour, because the watch doesn't measure the heart rate when the screen is off. This means that if it displayed the last value, it could be very out of date. There is an issue for measuring the heart rate (periodically) when the display is off: #183.

closed 2023-06-28 by FintasticMan

ajira86, 2023-06-29:
Thanks for your answer. I forget to check the closed issues. It could be nice to change zero value by a question mark or something that explain this situation.

> 15 seconds seems to be enough, but there is a further issue: The HR display on the home screen resets when the display is turned off. So if the HR is calculated thirteen seconds into the timeout, the measurement is discarded even if the user turns on the display a second after the timeout.
> 
> I guess the best option is to just remember the last heart rate measured and mark it as "stale" (maybe colour it yellow) instead of just showing 0. Of course none of this would be necessary if the HR sensor worked while the display is turned off.

The suggestion of changing the colour when value is old is an interesting way but the ticket was closed.