InfiniTimeOrg/InfiniTime #1789: 536876864 steps on the counter

Knirps1, 2023-07-01:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

The step counter on the watchface suddenly displayed 536876864 steps.

### What should happen instead?

The step counter should have displayed a few thousand steps.

### Reproduction steps

I don't know how to trigger this bug.

### More details?

I was dancing and afterwards the watch displayed this absurd step count. It could possibly be 2^29 plus around 6000 steps I had already taken that day.
The number on the watch did not change with further steps and stayed the same past midnight. The number got sent to Gadgetbridge and continued to go up there with further steps I took. I haven't restarted the watch since and held off on applying the update to 1.13.0

[image]
[image]
[image]
[image]
[image]
[image]

### Version

v1.12.0

### Companion app

Gadgetbridge 0.74.0

Boteium, 2023-07-01:
probably dead sensor
It's quite common
for example, drop the watch on the floor can cause this

Knirps1, 2023-07-05:
This appears to be it. The watch did fall  roughly 90 minutes before I saw the absurd count. I'm still unsure, how the step count in GadgetBridge continued to rise, but I it stays at 0 now after I restarted the watch.

closed 2023-07-05 by Knirps1