InfiniTimeOrg/InfiniTime #1790: Battery rapidly draining

segaretroboy, 2023-07-02:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

I went to bed with the watch near half charged, woke up with no charge.

### What should happen instead?

I was expecting the watch to make it through the night.

### Reproduction steps

I have no idea how this happened.

### More details?

Gadgetbridge shows the battery dropped rapidly at around 5:30AM

[image]
[image]

### Version

V1.13

### Companion app

GadgetBridge 0.74.0

steinhmor, 2023-07-04:
Same problem here. With the Update to 1.13 my battery only last half a day compared to 3,5 days before the update.
[image]

FintasticMan, 2023-07-04:
I think these are different issues. I think @segaretroboy's issue is somehow caused by a bug causing some component (like the display) to stay on when it should turn off.

I think that @steinhmor's issue is something else, but I'm not sure what. Could you tell me what features you've enabled (such as heart rate, raise to wake etc.) and what your display timeout is set to? I personally got just under a week of usage before the changes in 1.13, and get over 2 weeks with those changes, so it's really surprising to see such short battery life.

segaretroboy, 2023-07-10:
@FintasticMan I think you are correct on my issue, that was my suspicion but unfortunately I couldn't tell which it was since it happened overnight, the watch was acting normal the night prior. 
When I took it off the screen kept trying to turn on and I had to fiddle with the button, but it didn't come back on before I walked away from it after setting it down so that was probably unrelated. The watch has been acting normal since this as well.
[image]

steinhmor, 2023-07-10:
My problem also seems to be solved. After checking and setting the shake calibration (which might have been set to 0), my pinetime-battery seems to last muuuuuch longer. Thanks for the help!

JF002, 2023-08-06:
Thanks for the update, I guess we can close this issue, then.

segaretroboy, 2023-11-22:
Update on this, I recently had the watch do this again, after taking it off the charger, with nothing unusual with the watch's operation (screen was not stuck on and nothing was misbehaving) the watch went through 20% of the battery in the first 20 minutes off the charger, after this it re-stabilized. But it went from 61% to 42% between 1541 and 1600
Images are attached below
[image]
[image]
[image]

FintasticMan, 2023-11-22:
That's expected, the battery percentage displayed while charging isn't accurate and will always be higher than what it should be. After taking it off the charger, it'll update to the correct percentage within 10 mins. It's best to leave it on the charger until it says 100%, in which case it should be fully charged.

segaretroboy, 2024-02-08:
Not sure what happened this time, but this seems to have reoccurred in 1.14, went to bed around 3, and at 7AM my battery tanked, where I found the watch, it was face up, on a flat surface, with nothing near the button.
[image]
[image]
Only wake option selected is shake wake and the screen timeout is 7 seconds.

segaretroboy, 2024-02-08:
Actually it went dead in 24hrs
[image]
[image]
Off charger nearly full at 1430ish, dead by 1300ish

FintasticMan, 2024-02-08:
There does seem to be an issue here, so I'll reopen this. It's very unfortunate that it happens infrequently enough that it's very hard to figure out what's causing it. Had it completely died by the time you checked it in the morning? If it hadn't, did you notice anything like the screen being on?

segaretroboy, 2024-02-08:
I don't think the screen was on, I was up at around 12 today and didn't notice it if it was. I noticed when I put it on, shook it a few times, and pressed the button getting no response from it. I'll see if I cant set it in front of a webcam to see if I can catch the next time the issue appears.

segaretroboy, 2024-03-07:
Update on this, the issue occurred while the watch was on my wrist at work. The screen ceased dimming or turning off, the touchscreen completely stopped recognizing input, and the button was limited in what it could do, holding the button successfully brought up the system info screen, and double tapping let me see the last notification, single pressing the button also successfully returned me to the time screen, but I could not scroll in any of these since the touch screen wasn't responding. I was able to restore normal function by rebooting by holding the button down.
From 17:04 to 17:45 the screen did not turn off.
[image]

segaretroboy, 2024-03-07:
> That's useful information! It sounds very similar to what was reported here: [#2012 (comment)](https://github.com/InfiniTimeOrg/InfiniTime/issues/2012#issuecomment-1977602996).

Not sure if a notification triggered it, but that does look related, pretty much the same symptoms.

segaretroboy, 2024-04-05:
Update on this, caught the screen stuck on last night, but this time pressing the button to put it to sleep was all that was needed to restore normal function, it seems the screen staying on and the input suddenly stopping might be separate issues that just often come together.

segaretroboy, 2024-06-08:
One extra update, I have caught the watch doing it a few times now and I've noticed if I use the power button to switch the screen off, it does work, but the next time the screen comes on, it will still be effected by this bug still, but setting the screen timeout to something different and immediately back seems to persistently fix it, so I suspect the screen timeout value in RAM is getting overwritten or corrupted somehow.

JF002, 2024-06-09:
I also found my watch  sitting on a table with the display ON while it should have shut down for a while. I haven't had the opportunity to reproduce and diagnose this issue so far, though.
But yeah, the watch will definitively use a lot of power if it sits with the display ON for a long time (iirc, it should last for less that a day with the display on).

mark9064, 2024-06-09:
I've also experienced this, but I've got a lot of patches on top of main so I don't know if it's caused by changes I've made

Limited testing so far, which I'm not 100% sure on. Someone else should verify this:
Using just the button to sleep the watch and wake it again causes it to still stay on. A swipe left and back right is enough for the sleep timer to begin working again. The next time I see this I'll test if a tap is enough.

Either way it's looking like the LVGL activity timer has an incorrect value as displayapp regenerates IsPastSleepTime every refresh cycle, and that relies on only the settings controller and `lv_disp_get_inactive_time` (and a swipe left and right shouldn't affect the settings controller)

mark9064, 2024-06-11:
Hit this again just now. Turned on and off multiple times using the button and raise-wake/lower-sleep, screen timeout still broken. One tap on the screen, and exactly 7s (the configured timeout) later it went to sleep. So does seem like an activity timer issue

mark9064, 2025-06-28:
This should be resolved by the above 2 PRs