InfiniTimeOrg/InfiniTime #1801: infinitime not exposing BLE characteristics for the filesystem

dozadoesit, 2023-07-14:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

the filesystem cannot be interacted with to load resources

### What should happen instead?

the UUID for the filesystem should be exposed so resources can be loaded

### Reproduction steps

I installed infinitime 1.13.0 from gadgetbridge after that when connecting my watch to the computer over bluetooth I am not able to interact with any of the file system elements.  

### More details?

I contacted the dev for ITD and opened up  a ticket : [itd ticket 64](https://gitea.elara.ws/Elara6331/itd/issues/64)
We were unable to resolve the issues in that ticket because the watch is not exposing the file system over bluetooth.  doing some additional testing I tried to reflash the watch with 1.13.0 and that still didn't work. I also tried to unpair and re-pair to my computer and that also didn't work. 

Looking at the bluetooth information I see this 

Device CC:85:5F:53:88:F1 (random)
	Name: InfiniTime
	Alias: InfiniTime
	Appearance: 0x00c2
	Paired: yes
	Trusted: no
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Vendor specific           (00000000-78fc-48fe-8e23-433b3a1942d0)
	UUID: Vendor specific           (00001530-1212-efde-1523-785feabcd123)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Immediate Alert           (00001802-0000-1000-8000-00805f9b34fb)
	UUID: Current Time Service      (00001805-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: Heart Rate                (0000180d-0000-1000-8000-00805f9b34fb)
	UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
	UUID: Alert Notification Serv.. (00001811-0000-1000-8000-00805f9b34fb)
	UUID: adafruit industries       (0000febb-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (00010000-78fc-48fe-8e23-433b3a1942d0)
	UUID: Vendor specific           (00030000-78fc-48fe-8e23-433b3a1942d0)
	UUID: Vendor specific           (00040000-78fc-48fe-8e23-433b3a1942d0)

according to the developer of ITD there should be a UUID of adaf0200-4669-6c65-5472-616e73666572 but I don't see that listed anywhere. To date I've still not been able to load any resources to the watch or interact with the filesystem. 

### Version

1.13.0

### Companion app

ITD version 1.1.0

Avamander, 2023-07-14:
Try unpairing and re-pairing, your PC seems to be caching older characteristics.

dozadoesit, 2023-07-15:
As far as I know using bluetoothctl should expose all the bluetooth characteristics unless I am missing something. I did try to unpair and re-pair by running bluetoothctl remove <MAC Address> then bluetoothctl untrust <MAC Address> after I did  that I ran systemctl --user start itd and then put in the pair code that was shown on the watch. I am not sure if there is another way to unpair and re-pair my watch that I don't know about? after trying these steps I tried again to load the resource package and it didn't work.

Avamander, 2023-07-22:
That is quite certainly just the list of services, not all the characteristics. In any case, there's simply no way this can happen only for you but nobody else, so it must be something in your software (or firmware) stack.

FintasticMan, 2023-08-20:
I will close this issue, since it most likely isn't an InfiniTime issue.

closed 2023-08-20 by FintasticMan