InfiniTimeOrg/InfiniTime #1807: Docker build failing ("Cannot find source file:")

yblainm, 2023-07-15:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

I tried to follow the instructions at https://github.com/patricgruber/InfiniTime/blob/heartrate-measurements-in-background/doc/buildWithDocker.md (whether I build the image myself or not) but it fails because it can't find various libs/... files

### Reproduction steps

- Checkout 1.13.0 (I was originally trying a PR I was interested in using).
- Run `docker build -t infinitime-build ./docker`
- Run `docker run --rm -it -v ${PWD}:/sources --user $(id -u):$(id -g) infinitime-build /opt/build.sh pinetime-app`

I also tried `((HEAD detached at 1.13.0))$ docker run --rm -it -v ${PWD}:/sources --user $(id -u):$(id -g) infinitime/infinitime-build` and got the same

```
CMake Error at src/CMakeLists.txt:883 (add_library):
  Cannot find source file:

    libs/lvgl/lvgl.h

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .mpp .m .M .mm .ixx .cppm .h
  .hh .h++ .hm .hpp .hxx .in .txx .f .F .for .f77 .f90 .f95 .f03 .hip .ispc
```

### More details?

Here's what I ran and the output I got. To highlight:
```
CMake Error at src/CMakeLists.txt:883 (add_library):
  Cannot find source file:

    libs/lvgl/lvgl.h
```
I note that when I run the image with bash and look in `/sources/src/libs/lvgl/`, it's empty. I'm not sure if there's a dependency that I need to provide outside or if the Dockerfile is missing something.

```
((HEAD detached at 1.13.0))$ docker run --rm -it -v ${PWD}:/sources --user $(id -u):$(id -g) infinitime-build /opt/build.sh pinetime-app
+ set -e
+ export TOOLS_DIR=/opt
+ TOOLS_DIR=/opt
+ export SOURCES_DIR=/sources
+ SOURCES_DIR=/sources
+ export BUILD_DIR=/sources/build
+ BUILD_DIR=/sources/build
+ export OUTPUT_DIR=/sources/build/output
+ OUTPUT_DIR=/sources/build/output
+ export NPM_DIR=/sources/build/npm
+ NPM_DIR=/sources/build/npm
+ export npm_config_cache=/sources/build/npm
+ npm_config_cache=/sources/build/npm
+ export BUILD_TYPE=Release
+ BUILD_TYPE=Release
+ export GCC_ARM_VER=10.3-2021.10
+ GCC_ARM_VER=10.3-2021.10
+ export NRF_SDK_VER=nRF5_SDK_15.3.0_59ac345
+ NRF_SDK_VER=nRF5_SDK_15.3.0_59ac345
++ uname -m
+ MACHINE=x86_64
+ '[' x86_64 = arm64 ']'
+ export GCC_ARM_PATH=gcc-arm-none-eabi-10.3-2021.10
+ GCC_ARM_PATH=gcc-arm-none-eabi-10.3-2021.10
+ '[' false = false ']'
+ main pinetime-app
+ local target=pinetime-app
+ mkdir -p /opt
+ '[' '!' -d /opt/gcc-arm-none-eabi-10.3-2021.10 ']'
+ '[' '!' -d /opt/nRF5_SDK_15.3.0_59ac345 ']'
+ '[' '!' -d /opt/mcuboot ']'
+ mkdir -p /sources/build
+ CmakeGenerate
+ cmake -G 'Unix Makefiles' -S /sources -B /sources/build -DCMAKE_BUILD_TYPE=Release -DARM_NONE_EABI_TOOLCHAIN_PATH=/opt/gcc-arm-none-eabi-10.3-2021.10 -DNRF5_SDK_PATH=/opt/nRF5_SDK_15.3.0_59ac345 -DBUILD_DFU=1 -DBUILD_RESOURCES=1
PROJECT_GIT_COMMIT_HASH_SUCCESS? 0

BUILD CONFIGURATION
-------------------
    * Mode : Release
    * Version : 1.13.0
    * Toolchain : /opt/gcc-arm-none-eabi-10.3-2021.10
    * GitRef(S) : 5f3acfad
    * NRF52 SDK : /opt/nRF5_SDK_15.3.0_59ac345
    * Target device : PINETIME
    * Build DFU (using adafruit-nrfutil) : Enabled
    * Build resources : Enabled
-- Using /usr/bin/lv_font_conv to generate font files
-- Using /usr/bin/lv_font_conv to generate font files
-- Using /usr/bin/lv_img_conv to generate font files
-- Configuring done
CMake Error at src/CMakeLists.txt:883 (add_library):
  Cannot find source file:

    libs/lvgl/lvgl.h

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .mpp .m .M .mm .ixx .cppm .h
  .hh .h++ .hm .hpp .hxx .in .txx .f .F .for .f77 .f90 .f95 .f03 .hip .ispc

CMake Error at src/CMakeLists.txt:895 (add_library):
  Cannot find source file:

    libs/QCBOR/src/ieee754.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .mpp .m .M .mm .ixx .cppm .h
  .hh .h++ .hm .hpp .hxx .in .txx .f .F .for .f77 .f90 .f95 .f03 .hip .ispc

CMake Error at src/CMakeLists.txt:916 (add_library):
  Cannot find source file:

    libs/littlefs/lfs_util.h

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .mpp .m .M .mm .ixx .cppm .h
  .hh .h++ .hm .hpp .hxx .in .txx .f .F .for .f77 .f90 .f95 .f03 .hip .ispc

CMake Error at src/CMakeLists.txt:883 (add_library):
  No SOURCES given to target: lvgl

CMake Error at src/CMakeLists.txt:895 (add_library):
  No SOURCES given to target: QCBOR

CMake Error at src/CMakeLists.txt:916 (add_library):
  No SOURCES given to target: littlefs
```

### Version

1.13.0

yblainm, 2023-07-15:
I git cloned the various dependencies' sources by googling (for lvgl I needed to checkout `releases/v7`). Running `/opt/build.sh` now passes the above errors, but fails at the following:

```/sources/src/displayapp/LittleVgl.cpp:138:7: error: 'lv_disp_set_direction' was not declared in this scope; did you mean 'lv_disp_set_rotation'?
  138 |       lv_disp_set_direction(lv_disp_get_default(), 1);
      |       ^~~~~~~~~~~~~~~~~~~~~
      |       lv_disp_set_rotation
```
(and other such errors) followed by
```
gmake[2]: *** [src/CMakeFiles/pinetime-mcuboot-app.dir/build.make:1546: src/CMakeFiles/pinetime-mcuboot-app.dir/displayapp/LittleVgl.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
/sources/src/components/heartrate/Ppg.cpp: In member function 'int Pinetime::Controllers::Ppg::ProcessHeartRate(bool)':
/sources/src/components/heartrate/Ppg.cpp:203:3: error: 'ArduinoFFT' was not declared in this scope; did you mean 'arduinoFFT'?
  203 |   ArduinoFFT<float> FFT = ArduinoFFT<float>(vReal.data(), vImag.data(), dataLength, sampleFreq);
      |   ^~~~~~~~~~
      |   arduinoFFT
/sources/src/components/heartrate/Ppg.cpp:203:14: error: expected primary-expression before 'float'
  203 |   ArduinoFFT<float> FFT = ArduinoFFT<float>(vReal.data(), vImag.data(), dataLength, sampleFreq);
      |              ^~~~~
/sources/src/components/heartrate/Ppg.cpp:204:3: error: 'FFT' was not declared in this scope
  204 |   FFT.compute(FFTDirection::Forward);
      |   ^~~
/sources/src/components/heartrate/Ppg.cpp:206:18: error: expected class-name before '(' token
  206 |   FFT.~ArduinoFFT();
      |                  ^
gmake[2]: *** [src/CMakeFiles/pinetime-mcuboot-app.dir/build.make:1644: src/CMakeFiles/pinetime-mcuboot-app.dir/components/heartrate/Ppg.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:329: src/CMakeFiles/pinetime-mcuboot-app.dir/all] Error 2
gmake: *** [Makefile:91: all] Error 2
```
I assumed this is due to me providing the wrong versions of the dependencies.

Avamander, 2023-07-15:
Make sure to just clone the repo with submodules, or initialize them afterwards. It's inadvisable to start and manually find all the dependencies.

yblainm, 2023-07-15:
To be clear, I should clone the repository using `git clone --recurse-submodules`, right? That appears to have been the problem.

Apologies for my ignorance. For future reference, is there anywhere I would look to know that in advance? I suppose there is a .gitmodules, after all.

closed 2023-07-15 by yblainm

FintasticMan, 2023-07-15:
It mentions running `git submodule update --init` in `doc/buildAndProgram.md`, perhaps that should also be added to the Docker build instructions?