InfiniTimeOrg/InfiniTime #1813: Heart rate measurement is slow

everypizza1, 2023-07-20:
### Verification

- [X] I searched for similar issues and found none was relevant.

### Introduce the issue

My heart rate sensor is giving correct info (around 100 BPM) but takes a long time.

Whenever I start it, it takes ~1 minute to start and when I move my arm, it takes another minute. 

### Preferred solution

I'm not a cardiologist so I have no clue how to solve this. 

### Version

1.13.0

minacode, 2023-07-21:
I measured and got 8-9 seconds. Could it be that it is not able to get good values and keeps on trying? Do you have stable output once you got your first value or does it go back to zero sometimes?

everypizza1, 2023-07-21:
It's stable until I move my arm. It might be worth noting that I am 13.

everypizza1, 2023-07-22:
I realized that there was still protective film over the heart rate sensor. It now takes less time to start measuring.

closed 2023-07-22 by everypizza1