InfiniTimeOrg/InfiniTime #1815: Missing Widget?

mattklunder, 2023-07-21:
### Verification

- [X] I searched for similar issues and found none was relevant.

### Introduce the issue

In my menu of programs, i have one screen with 6 widgets (stopwatch, alarm, timer, steps, heart rate, music) and a second screen with just 5 (pen/drawing, pong, 2048, metronome, navigation).  While I can't remember what it was, I could swear there used to be a sixth widget there.  Am I missing something that I should expect to see?

### Version

1.13.0

everypizza1, 2023-07-21:
I found out a few days ago that it was an accelerator data page. No clue why it was there.

mattklunder, 2023-07-21:
Interesting, thanks for that!  Do you see the same mix of widgets that 
I do?

everypizza1, 2023-07-21:
Yes.

mattklunder, 2023-07-21:
closing this since what I'm observing appears to be expected, thanks!

closed 2023-07-21 by mattklunder

FintasticMan, 2023-07-22:
The motion app was removed in #1603.