InfiniTimeOrg/InfiniTime #1833: InfiniTime wont build

hexisXz, 2023-08-16:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

i ran sudo make -j16 pinetime-mcuboot-app and it wont build 

### What should happen instead?

it should build 

### Reproduction steps

i rant the command sudo make -j16 pinetime-mcuboot-app and i get these errors 

[  0%] Built target infinitime_fonts_jetbrains_mono_76
[  0%] Built target infinitime_fonts_fontawesome_weathericons
[  0%] Built target infinitime_fonts_jetbrains_mono_bold_20
[  3%] Built target infinitime_fonts_jetbrains_mono_42
[  3%] Built target infinitime_fonts_lv_font_navi_80
[  3%] Built target infinitime_fonts_open_sans_light
[  3%] Built target infinitime_fonts_lv_font_sys_48
[  3%] Built target infinitime_fonts_jetbrains_mono_extrabold_compressed
[  3%] Building C object src/CMakeFiles/QCBOR.dir/libs/QCBOR/src/qcbor_encode.c.o
[  3%] Building C object src/CMakeFiles/QCBOR.dir/libs/QCBOR/src/ieee754.c.o
[  3%] Building C object src/CMakeFiles/QCBOR.dir/libs/QCBOR/src/qcbor_decode.c.o
[  3%] Building C object src/CMakeFiles/littlefs.dir/libs/littlefs/lfs.c.o
[  3%] Building C object src/CMakeFiles/littlefs.dir/libs/littlefs/lfs_util.c.o
[  3%] Building C object src/CMakeFiles/nimble.dir/libs/mynewt-nimble/porting/npl/freertos/src/nimble_port_freertos.c.o
/bin/sh: 1: /home/stuff/bin/arm-none-eabi-gcc: not found
/bin/sh: 1: /home/stuff/bin/arm-none-eabi-gcc: not found
[  3%] Building C object src/CMakeFiles/nrf-sdk.dir/home/hexis/Downloads/nRF5_SDK_15.3.0_59ac345/modules/nrfx/mdk/system_nrf52.c.o
[  3%] Building C object src/CMakeFiles/lvgl.dir/libs/lvgl/src/lv_core/lv_disp.c.o
make[3]: *** [src/CMakeFiles/QCBOR.dir/build.make:76: src/CMakeFiles/QCBOR.dir/libs/QCBOR/src/ieee754.c.o] Error 127
make[3]: *** Waiting for unfinished jobs....
make[3]: *** [src/CMakeFiles/QCBOR.dir/build.make:104: src/CMakeFiles/QCBOR.dir/libs/QCBOR/src/qcbor_encode.c.o] Error 127
/bin/sh: 1: /home/stuff/bin/arm-none-eabi-gcc: not found
make[3]: *** [src/CMakeFiles/QCBOR.dir/build.make:90: src/CMakeFiles/QCBOR.dir/libs/QCBOR/src/qcbor_decode.c.o] Error 127
/bin/sh: 1: /home/stuff/bin/arm-none-eabi-gcc: not found
make[3]: *** [src/CMakeFiles/littlefs.dir/build.make:90: src/CMakeFiles/littlefs.dir/libs/littlefs/lfs.c.o] Error 127
make[3]: *** Waiting for unfinished jobs....
/bin/sh: 1: /home/stuff/bin/arm-none-eabi-gcc: not found
/bin/sh: 1: /home/stuff/bin/arm-none-eabi-gcc: not found
make[3]: *** [src/CMakeFiles/littlefs.dir/build.make:76: src/CMakeFiles/littlefs.dir/libs/littlefs/lfs_util.c.o] Error 127
make[3]: *** [src/CMakeFiles/nrf-sdk.dir/build.make:76: src/CMakeFiles/nrf-sdk.dir/home/hexis/Downloads/nRF5_SDK_15.3.0_59ac345/modules/nrfx/mdk/system_nrf52.c.o] Error 127
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/Makefile2:247: src/CMakeFiles/littlefs.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
/bin/sh: 1: /home/stuff/bin/arm-none-eabi-gcc: not found
make[3]: *** [src/CMakeFiles/lvgl.dir/build.make:76: src/CMakeFiles/lvgl.dir/libs/lvgl/src/lv_core/lv_disp.c.o] Error 127
make[3]: *** Waiting for unfinished jobs....
[  3%] Building C object src/CMakeFiles/nimble.dir/libs/mynewt-nimble/nimble/host/src/ble_hs.c.o
[  3%] Building C object src/CMakeFiles/lvgl.dir/libs/lvgl/src/lv_core/lv_group.c.o
[  3%] Building C object src/CMakeFiles/QCBOR.dir/libs/QCBOR/src/qcbor_err_to_str.c.o
[  3%] Building ASM object src/CMakeFiles/nrf-sdk.dir/home/hexis/Downloads/nRF5_SDK_15.3.0_59ac345/modules/nrfx/mdk/gcc_startup_nrf52.S.o
[  3%] Building C object src/CMakeFiles/nimble.dir/libs/mynewt-nimble/porting/npl/freertos/src/npl_os_freertos.c.o
[  5%] Building C object src/CMakeFiles/nrf-sdk.dir/home/hexis/Downloads/nRF5_SDK_15.3.0_59ac345/components/boards/boards.c.o
/bin/sh: 1: /home/stuff/bin/arm-none-eabi-gcc: not found
make[3]: *** [src/CMakeFiles/lvgl.dir/build.make:90: src/CMakeFiles/lvgl.dir/libs/lvgl/src/lv_core/lv_group.c.o] Error 127
/bin/sh: 1: /home/stuff/bin/arm-none-eabi-gcc: not found
/bin/sh: 1: /home/stuff/bin/arm-none-eabi-gcc: not found
make[3]: *** [src/CMakeFiles/QCBOR.dir/build.make:118: src/CMakeFiles/QCBOR.dir/libs/QCBOR/src/qcbor_err_to_str.c.o] Error 127
make[3]: *** [src/CMakeFiles/nimble.dir/build.make:76: src/CMakeFiles/nimble.dir/libs/mynewt-nimble/porting/npl/freertos/src/nimble_port_freertos.c.o] Error 127
make[2]: *** [CMakeFiles/Makefile2:221: src/CMakeFiles/QCBOR.dir/all] Error 2
make[3]: *** Waiting for unfinished jobs....
[  7%] Building C object src/CMakeFiles/lvgl.dir/libs/lvgl/src/lv_core/lv_indev.c.o
/bin/sh: 1: /home/stuff/bin/arm-none-eabi-gcc: not found[  7%] Building C object src/CMakeFiles/nimble.dir/libs/mynewt-nimble/nimble/host/src/ble_hs_hci_evt.c.o

[  7%] Building C object src/CMakeFiles/lvgl.dir/libs/lvgl/src/lv_core/lv_obj.c.o
[... 38 lines cut ...]
/bin/sh: 1: /home/stuff/bin/arm-none-eabi-gcc: not found
/bin/sh: 1: /home/stuff/bin/arm-none-eabi-gcc: not found
/bin/sh: 1: /home/stuff/bin/arm-none-eabi-gcc: not found
[ 10%] Building C object src/displayapp/fonts/CMakeFiles/infinitime_fonts.dir/fontawesome_weathericons.c.o
/bin/sh: 1: /home/stuff/bin/arm-none-eabi-gcc: not found
/bin/sh: 1: /home/stuff/bin/arm-none-eabi-gcc: not found
make[3]: *** [src/displayapp/fonts/CMakeFiles/infinitime_fonts.dir/build.make:192: src/displayapp/fonts/CMakeFiles/infinitime_fonts.dir/open_sans_light.c.o] Error 127
make[3]: *** Waiting for unfinished jobs....
make[3]: *** [src/displayapp/fonts/CMakeFiles/infinitime_fonts.dir/build.make:108: src/displayapp/fonts/CMakeFiles/infinitime_fonts.dir/jetbrains_mono_42.c.o] Error 127
make[3]: *** [src/displayapp/fonts/CMakeFiles/infinitime_fonts.dir/build.make:122: src/displayapp/fonts/CMakeFiles/infinitime_fonts.dir/jetbrains_mono_76.c.o] Error 127
make[3]: *** [src/displayapp/fonts/CMakeFiles/infinitime_fonts.dir/build.make:136: src/displayapp/fonts/CMakeFiles/infinitime_fonts.dir/jetbrains_mono_bold_20.c.o] Error 127
make[3]: *** [src/displayapp/fonts/CMakeFiles/infinitime_fonts.dir/build.make:150: src/displayapp/fonts/CMakeFiles/infinitime_fonts.dir/jetbrains_mono_extrabold_compressed.c.o] Error 127
/bin/sh: 1: /home/stuff/bin/arm-none-eabi-gcc: not found
make[3]: *** [src/displayapp/fonts/CMakeFiles/infinitime_fonts.dir/build.make:164: src/displayapp/fonts/CMakeFiles/infinitime_fonts.dir/lv_font_navi_80.c.o] Error 127
/bin/sh: 1: /home/stuff/bin/arm-none-eabi-gcc: not found
make[3]: *** [src/displayapp/fonts/CMakeFiles/infinitime_fonts.dir/build.make:178: src/displayapp/fonts/CMakeFiles/infinitime_fonts.dir/lv_font_sys_48.c.o] Error 127
/bin/sh: 1: /home/stuff/bin/arm-none-eabi-gcc: not found
make[3]: *** [src/displayapp/fonts/CMakeFiles/infinitime_fonts.dir/build.make:206: src/displayapp/fonts/CMakeFiles/infinitime_fonts.dir/fontawesome_weathericons.c.o] Error 127
make[2]: *** [CMakeFiles/Makefile2:460: src/displayapp/fonts/CMakeFiles/infinitime_fonts.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:316: src/CMakeFiles/pinetime-mcuboot-app.dir/rule] Error 2
make: *** [Makefile:202: pinetime-mcuboot-app] Error 2

### Version

1.13.0

Avamander, 2023-08-16:
You've configured your build environment incorrectly. Make sure you follow the build guide and define the compiler's location correctly.

closed 2023-08-16 by Avamander

FintasticMan, 2023-08-16:
You really shouldn't use sudo when running any build tool, it will just mess up the permissions of the generated files.