InfiniTimeOrg/InfiniTime #1839: False touches create chaos on my watch

WhyNotHugo, 2023-08-21:
### Verification

- [X] I searched for similar issues (including closed issues) and found none was relevant.

### Introduce the issue

There's a lot of situations where my Pinetime registers false taps/drags:

- When the material of the inside of my coat touches the screen (one of the coats looks like the interior material is the same as those gloves that work on touch screens, but it happens with both coats).
- If I fall asleep with my watch on, it'll touch my arm or whatever as I move around
- When somebody is next to me and the watch screen touches them, that also registers
- I think this might also happen when there's rain? Not 100% sure about this particular one

The practical results of these issues, is that I often lift my arm to look at my watch, and I suddenly have a new watch face (because accidental taps reached that menu and changed the configured face), or sometimes a game of pong is running, or some random menu is open. I've see it all so far.

I have "wake up" set to "single tap" and "raise wrist" (actually, I just checked and these settings seem to have gotten garbled last time I used the watch, but these are my _intended_ settings).

I could disabled these to prevent the issue, but that would also mean that I'd need both hands unlock the watch and look at the time: definitely not convenient.

### Preferred solution

I think the most useful approach would be to split the "wake up" setting from the "unlock" setting.

I want my watch to wake up (e.g.: the screen to power on) when I tap once or raise my arm, but I don't want the screen to unlock (e.g.: register taps).

I think leaving a separate setting for "unlock" would be useful, since I'd leave all boxes unchecked for that and require touching the side button to unlock it.

### Version

1.13.0 (this has happened since forever)

WhyNotHugo, 2023-08-21:
https://github.com/InfiniTimeOrg/InfiniTime/pull/961 is somewhat related: it tries to address one of the problematic scenarios with a different approach.

However, looks like that patchset won't be merged, the solution proposed here is different and it covers other scenarios too.

hexisXz, 2023-08-21:
You could just use raise to wake, shake or double tap to wake.

FintasticMan, 2023-08-21:
My pull request that was merged recently significantly reduces the false positives in the raise wrist algorithm.

KaffeinatedKat, 2023-09-30:
#1359 addresses this issue, i've been using this for a while and it works great