InfiniTimeOrg/InfiniTime #1842: Short ref doesn't appear in CI builds

Commenter25, 2023-08-25:
### Verification

- [X] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

When checking the About or Firmware pages, if running a build from GitHub CI, there will be no short ref

### Reproduction steps

1. Download and install a [release build](https://github.com/InfiniTimeOrg/InfiniTime/releases)
2. Go to the about page and see a commit reference next to short ref (for 1.13, this is `5f3acfad`)
3. Download and install a [CI build](https://github.com/InfiniTimeOrg/InfiniTime/actions/runs/5364794676) (this is the 1.13 commit)
4. Go to the about page and notice a distinct lack of a short ref

### Version

v1.13.0

jensonb, 2023-09-02:
I also noticed this issue as well when building myself. Where does the release build actually come from? I'm a little saddened to see that it is not the build from the CI action 🙁.

Either way, the git hash is retrieved during the following portion of the build:
https://github.com/InfiniTimeOrg/InfiniTime/blob/5f3acfadd24fcf8f0ef31a75c36180585b1c88c4/CMakeLists.txt#L38-L57
https://github.com/InfiniTimeOrg/InfiniTime/actions/runs/5364794676/job/14523852337#step:4:55
[image]

the git command seems to be failing

FintasticMan, 2024-11-18:
Seems that PR didn't actually fix the issue in CI.

PR #1935 (2024-11-17): Fix git hash calculation with Docker build
Fix error when compiling the Pinetime using the Docker image. If done with Docker, the container does not trust the /sources folder, leading to a blank response of the command that grabs the git commit `git rev-parse --short HEAD`.

```
fatal: detected dubious ownership in repository at '/sources'
To add an exception for this directory, call:

        git config --global --add safe.directory /sources
PROJECT_GIT_COMMIT_HASH_SUCCESS? 128

BUILD CONFIGURATION
-------------------
    * Mode : Release
    * Version : 1.3.0
    * Toolchain : /opt/gcc-arm-n [... cut]