InfiniTimeOrg/InfiniTime #1843: Motion Controller issue when screen off

Dheenhasty, 2023-08-27:
### Verification

- [X] I searched for similar issues (including closed issues) and found none was relevant.

### Introduce the issue

Hello, 

    I'm not sure since when, but i have an issue regarding Bluetooth availability of Motion stats with "Sleep as Android", when screen is on, no issue motion is reported correctly, but when screen go to off, it doesn't report any more. 

    Not sure, if this behaviour is intended, or is it just me.

   Thank you for all your work !

### Version

v1.13.0

Dheenhasty, 2023-08-27:
Find a workarround, by activating the "shake to wake" or "raise wrist" wake up. 

seems like issue is only in tap, double tap or only button mode wake up. 

so it can be good to have a way to activate the background motion capture (just like heart rate maybe ? )

FintasticMan, 2023-08-27:
Oh yeah, forgot about this issue. It's fixed by #1207.

Dheenhasty, 2023-08-27:
i search for the same issue, but not this old :) thanks a lot

Great, so with that i will be able to follow my sleep

PR #1970 (2024-01-23): Fix motion service not updating when sleeping
This PR fixes the motion service not being updated when the device is sleeping. This would only happen in the case that there is no motion-based wake-up mode enabled.

Thanks to @apilat for the implementation.

Fixes #1843.
Supersedes #1207.
closed 2024-01-23 by FintasticMan