InfiniTimeOrg/InfiniTime #1846: Nodesource script is deprecated

blm768, 2023-08-31:
### Verification

- [X] I searched for similar issues (including closed issues) and found none was relevant.

### Introduce the issue

The dev container pulls in https://deb.nodesource.com/setup_18.x. Running that script produces a deprecation notice:/

```
================================================================================
▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓
================================================================================

                           SCRIPT DEPRECATION WARNING                    

  
  This script, located at https://deb.nodesource.com/setup_X, used to
  install Node.js is deprecated now and will eventually be made inactive.

  Please visit the NodeSource distributions ================================================================================
▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓
================================================================================

                           SCRIPT DEPRECATION WARNING                    

  
  This script, located at https://deb.nodesource.com/setup_X, used to
  install Node.js is deprecated now and will eventually be made inactive.

  Please visit the NodeSource distributions Github and follow the
  instructions to migrate your repo.
  https://github.com/nodesource/distributions

  The NodeSource Node.js Linux distributions GitHub repository contains
  information about which versions of Node.js and which Linux distributions
  are supported and how to install it.
  https://github.com/nodesource/distributions

                          SCRIPT DEPRECATION WARNING

================================================================================
▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓
================================================================================
Github and follow the
  instructions to migrate your repo.
  https://github.com/nodesource/distributions

  The NodeSource Node.js Linux distributions GitHub repository contains
  information about which versions of Node.js and which Linux distributions
  are supported and how to install it.
  https://github.com/nodesource/distributions

                          SCRIPT DEPRECATION WARNING

================================================================================
▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓▓
================================================================================
```

I don't know if this is related, but on my machine, the dev container is somehow missing `npm`. I'm not sure how the rest of the dev container can build if this step is failing, so that's pretty weird. My system might just be special, though. Running the script manually in the container does let me install `npm`.

### Version

v1.13.0

blm768, 2023-09-01:
Ah, I was looking at the wrong `Dockerfile`. That'd explain why the dev container didn't have NPM; it just wasn't installing it.

FintasticMan, 2023-09-01:
Yeah, the devcontainer is very outdated

blm768, 2023-09-03:
While it was up to date, did it have reasonable usability? I ask because I'm working on NixOS, which makes it kind of clunky to install all the dependencies via `apt`. There are various things I can try with the main `Dockerfile` and/or `systemd-machined` if the dev container was abandoned due to usability issues.

blm768, 2023-09-03:
Ooh, nice! I'll see if I can get things to build on top of that PR locally. If it looks like the PR's going to stall out, I could try to stack some fixes on it to address the outstanding issues.

JF002, 2023-12-10:
Fixed in #1849.

closed 2023-12-10 by JF002