InfiniTimeOrg/InfiniTime #1864: Watch face goes blank at times.

FNTONN, 2023-09-22:
### Verification

- [X] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

This specific watch face I like (Casio G7710) goes blank, shows no time or date.

### What should happen instead?

It should show the time and date when I look at it.

### Reproduction steps

After a certain period of time the watch would show a blank screen with the outlines but no time. I will upload a better photo later as I'm a bit busy at the moment. 
[image]

### Version

v1.13.0

### Companion app

InfiniLink for iOS 16.6 on iPhone 12 Mini

FNTONN, 2023-09-22:
[image][image]

mark9064, 2023-09-22:
You're probably experiencing #1411. There's a workaround there you can try if you're comfortable with compiling yourself

closed 2023-10-28 by FNTONN