InfiniTimeOrg/InfiniTime #1865: Weather icon in PTS not showing the correct weather

sonnatager, 2023-09-22:
### Verification

- [X] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

PTS not showing the correct weather icon

### What should happen instead?

Showing the correct weather icon

### Reproduction steps

1. Add different locations with different weathers to weather app
2. Switch between the locations (put the used location on top in Breezy Weather)
3. Temperature on the watch changes (but is not every time correct)
4. Icon is not correct

[image]

You can see, it is raining.
This is the watch displaying no rain:
[image]

Why is not the "Condition Code" used to decide which weather icon is displayed?
All conditions: https://openweathermap.org/weather-conditions

### More details?

Weather App: Breezy Weather

The cached weather data in Gadgetbridge looks like following:
[image]

You can see, it is raining.
This is the watch displaying no rain:
[image]

Why isn't the "Condition Code" used to decide which weather icon is displayed?
All conditions: https://openweathermap.org/weather-conditions

### Version

main

### Companion app

Gadgetbridge 0.76.0

Avamander, 2023-09-22:
Can you please provide Gadgetbridge's log when weather information is pushed?

sonnatager, 2023-09-22:
Here is the log of GB

`13:58:49.298 [main] INFO  n.f.g.e.GenericWeatherReceiver - Got generic weather for *****
13:58:49.299 [main] INFO  n.f.g.m.Weather - Loading weather from cache /data/user/0/nodomain.freeyourgadget.gadgetbridge/cache/weatherCache.bin
13:58:49.316 [main] DEBUG n.f.g.s.DeviceCommunicationService - Service startcommand: nodomain.freeyourgadget.gadgetbridge.devices.action.send_weather
13:58:49.318 [main] DEBUG n.f.g.s.b.BtLEQueue - about to add: 13:58:49: Transaction task: WeatherData with 1 actions
13:58:49.319 [Gadgetbridge GATT Dispatcher] DEBUG n.f.g.s.b.BtLEQueue - About to run action: 22 September, 13:58: WriteAction on characteristic: 00040001-78fc-48fe-8e23-433b3a1942d0
13:58:49.320 [Gadgetbridge GATT Dispatcher] DEBUG n.f.g.s.b.a.WriteAction - writing to characteristic: 00040001-78fc-48fe-8e23-433b3a1942d0: 0xbf 0x69 0x54 0x69 0x6d 0x65 0x73 0x74 0x61 0x6d 0x70 0x1a 0x65 0x0d 0x81 0x79 0x67 0x45 0x78 0x70 0x69 0x72 0x65 0x73 0x19 0x54 0x60 0x69 0x45 0x76 0x65 0x6e 0x74 0x54 0x79 0x70 0x65 0x07 0x68 0x4c 0x6f 0x63 0x61 0x74 0x69 0x6f 0x6e 0x67 0x4c 0x75 0x64 0x65 0x73 0x63 0x68 0x68 0x41 0x6c 0x74 0x69 0x74 0x75 0x64 0x65 0x00 0x68 0x4c 0x61 0x74 0x69 0x74 0x75 0x64 0x65 0x00 0x69 0x4c 0x6f 0x6e 0x67 0x69 0x74 0x75 0x64 0x65 0x00 0xff
13:58:49.322 [main] DEBUG n.f.g.s.b.BtLEQueue - about to add: 13:58:49: Transaction task: WeatherData with 1 actions
13:58:49.324 [main] DEBUG n.f.g.s.b.BtLEQueue - about to add: 13:58:49: Transaction task: WeatherData with 1 actions
13:58:49.326 [main] DEBUG n.f.g.s.b.BtLEQueue - about to add: 13:58:49: Transaction task: WeatherData with 1 actions
13:58:49.327 [main] DEBUG n.f.g.s.b.BtLEQueue - about to add: 13:58:49: Transaction task: WeatherData with 1 actions
13:58:49.328 [main] DEBUG n.f.g.s.b.BtLEQueue - about to add: 13:58:49: Transaction task: WeatherData with 1 actions
13:58:49.330 [main] DEBUG n.f.g.s.b.BtLEQueue - about to add: 13:58:49: Transaction task: WeatherData with 1 actions
13:58:49.332 [main] DEBUG n.f.g.s.b.BtLEQueue - about to add: 13:58:49: Transaction task: WeatherData with 1 actions
13:58:49.332 [main] DEBUG n.f.g.s.d.p.PineTimeJFSupport - Wrote weather data
13:58:49.395 [binder:30276_7] DEBUG n.f.g.s.b.BtLEQueue - characteristic write: 00040001-78fc-48fe-8e23-433b3a1942d0 (success)
13:58:49.401 [Gadgetbridge GATT Dispatcher] DEBUG n.f.g.s.b.BtLEQueue - About to run action: 22 September, 13:58: WriteAction on characteristic: 00040001-78fc-48fe-8e23-433b3a1942d0
13:58:49.406 [Gadgetbridge GATT Dispatcher] DEBUG n.f.g.s.b.a.WriteAction - writing to characteristic: 00040001-78fc-48fe-8e23-433b3a1942d0: 0xbf 0x69 0x54 0x69 0x6d 0x65 0x73 0x74 0x61 0x6d 0x70 0x1a 0x65 0x0d 0x81 0x79 0x67 0x45 0x78 0x70 0x69 0x72 0x65 0x73 0x19 0x54 0x60 0x69 0x45 0x76 0x65 0x6e 0x74 0x54 0x79 0x70 0x65 0x09 0x68 0x48 0x75 0x6d 0x69 0x64 0x69 0x74 0x79 0x18 0x4e 0xff
13:58:49.486 [binder:30276_7] DEBUG n.f.g.s.b.BtLEQueue - characteristic write: 00040001-78fc-48fe-8e23-433b3a1942d0 (success)
13:58:49.488 [Gadgetbridge GATT Dispatcher] DEBUG n.f.g.s.b.BtLEQueue - About to run action: 22 September, 13:58: WriteAction on characteristic: 00040001-78fc-48fe-8e23-433b3a1942d0
13:58:49.489 [Gadgetbridge GATT Dispatcher] DEBUG n.f.g.s.b.a.WriteAction - writing to characteristic: 00040001-78fc-48fe-8e23-433b3a1942d0: 0xbf 0x69 0x54 0x69 0x6d 0x65 0x73 0x74 0x61 0x6d 0x70 0x1a 0x65 0x0d 0x81 0x79 0x67 0x45 0x78 0x70 0x69 0x72 0x65 0x73 0x19 0x54 0x60 0x69 0x45 0x76 0x65 0x6e 0x74 0x54 0x79 0x70 0x65 0x03 0x6b 0x54 0x65 0x6d 0x70 0x65 0x72 0x61 0x74 0x75 0x72 0x65 0x19 0x04 0xa1 0x68 0x44 0x65 0x77 0x50 0x6f 0x69 0x6e 0x74 0x39 0x7f 0xff 0xff
13:58:49.575 [binder:30276_7] DEBUG n.f.g.s.b.BtLEQueue - characteristic write: 00040001-78fc-48fe-8e23-433b3a1942d0 (success)
13:58:49.576 [Gadgetbridge GATT Dispatcher] DEBUG n.f.g.s.b.BtLEQueue - About to run action: 22 September, 13:58: WriteAction on characteristic: 00040001-78fc-48fe-8e23-433b3a1942d0
13:58:49.577 [Gadgetbridge GATT Dispatcher] DEBUG n.f.g.s.b.a.WriteAction - writing to characteristic: 00040001-78fc-48fe-8e23-433b3a1942d0: 0xbf 0x69 0x54 0x69 0x6d 0x65 0x73 0x74 0x61 0x6d 0x70 0x1a 0x65 0x0d 0x81 0x79 0x67 0x45 0x78 0x70 0x69 0x72 0x65 0x73 0x1a 0x00 0x01 0x51 0x80 0x69 0x45 0x76 0x65 0x6e 0x74 0x54 0x79 0x70 0x65 0x03 0x6b 0x54 0x65 0x6d 0x70 0x65 0x72 0x61 0x74 0x75 0x72 0x65 0x19 0x05 0x69 0x68 0x44 0x65 0x77 0x50 0x6f 0x69 0x6e 0x74 0x39 0x7f 0xff 0xff
13:58:49.668 [binder:30276_7] DEBUG n.f.g.s.b.BtLEQueue - characteristic write: 00040001-78fc-48fe-8e23-433b3a1942d0 (success)
13:58:49.669 [Gadgetbridge GATT Dispatcher] DEBUG n.f.g.s.b.BtLEQueue - About to run action: 22 September, 13:58: WriteAction on characteristic: 00040001-78fc-48fe-8e23-433b3a1942d0
13:58:49.672 [Gadgetbridge GATT Dispatcher] DEBUG n.f.g.s.b.a.WriteAction - writing to characteristic: 00040001-78fc-48fe-8e23-433b3a1942d0: 0xbf 0x69 0x54 0x69 0x6d 0x65 0x73 0x74 0x61 0x6d 0x70 0x1a 0x65 0x0d 0x81 0x79 0x67 0x45 0x78 0x70 0x69 0x72 0x65 0x73 0x19 0x54 0x60 0x69 0x45 0x76 0x65 0x6e 0x74 0x54 0x79 0x70 0x65 0x02 0x68 0x53 0x70 0x65 0x65 0x64 0x4d 0x69 0x6e 0x02 0x68 0x53 0x70 0x65 0x65 0x64 0x4d 0x61 0x78 0x02 0x6c 0x44 0x69 0x72 0x65 0x63 0x74 0x69 0x6f 0x6e 0x4d 0x69 0x6e 0x18 0x70 0x6c 0x44 0x69 0x72 0x65 0x63 0x74 0x69 0x6f 0x6e 0x4d 0x61 0x78 0x18 0x70 0xff
13:58:49.756 [binder:30276_7] DEBUG n.f.g.s.b.BtLEQueue - characteristic write: 00040001-78fc-48fe-8e23-433b3a1942d0 (success)
[... cut ...]
13:58:49.944 [Gadgetbridge GATT Dispatcher] DEBUG n.f.g.s.b.a.WriteAction - writing to characteristic: 00040001-78fc-48fe-8e23-433b3a1942d0: 0xbf 0x69 0x54 0x69 0x6d 0x65 0x73 0x74 0x61 0x6d 0x70 0x1a 0x65 0x0d 0x81 0x79 0x67 0x45 0x78 0x70 0x69 0x72 0x65 0x73 0x19 0x54 0x60 0x69 0x45 0x76 0x65 0x6e 0x74 0x54 0x79 0x70 0x65 0x08 0x66 0x41 0x6d 0x6f 0x75 0x6e 0x74 0x18 0x4b 0xff
13:58:50.025 [binder:30276_7] DEBUG n.f.g.s.b.BtLEQueue - characteristic write: 00040001-78fc-48fe-8e23-433b3a1942d0 (success)
`

AgentConDier, 2023-10-03:
I've also never seen a different weather icon other than the :sun_behind_small_cloud:
My current condition code is 500 (light rain)
<details><summary>GadgetBridge log:</summary><pre><code>
21:10:07.494 [main] INFO  n.f.g.Logging - Gadgetbridge version: 0.76.1-660986910
21:10:07.496 [main] DEBUG n.f.g.a.AbstractPreferenceFragment - Preference changed: log_to_file

[…]

21:10:10.146 [main] DEBUG n.f.g.e.NotificationListener - Notification 0 posted: packageName=com.termux.api, priority=0, category=null
21:10:10.147 [main] INFO  n.f.g.e.NotificationListener - Allowing notification, application is whitelisted
21:10:10.147 [main] INFO  n.f.g.e.NotificationListener - Not forwarding notification, screen seems to be on and settings do not allow this
21:10:10.148 [main] INFO  n.f.g.e.NotificationListener - Ignoring notification: com.termux.api
21:10:11.506 [main] DEBUG n.f.g.s.DeviceCommunicationService - Service startcommand: nodomain.freeyourgadget.gadgetbridge.devices.action.send_weather
21:10:11.510 [main] DEBUG n.f.g.s.b.BtLEQueue - about to add: 21:10:11: Transaction task: WeatherData with 1 actions
21:10:11.511 [main] DEBUG n.f.g.s.b.BtLEQueue - about to add: 21:10:11: Transaction task: WeatherData with 1 actions
21:10:11.511 [main] DEBUG n.f.g.s.b.BtLEQueue - about to add: 21:10:11: Transaction task: WeatherData with 1 actions
21:10:11.511 [main] DEBUG n.f.g.s.b.BtLEQueue - about to add: 21:10:11: Transaction task: WeatherData with 1 actions
21:10:11.511 [main] DEBUG n.f.g.s.b.BtLEQueue - about to add: 21:10:11: Transaction task: WeatherData with 1 actions
21:10:11.511 [Gadgetbridge GATT Dispatcher] DEBUG n.f.g.s.b.BtLEQueue - About to run action: 3. Oktober, 21:10: WriteAction on characteristic: 00040001-78fc-48fe-8e23-433b3a1942d0
21:10:11.512 [main] DEBUG n.f.g.s.b.BtLEQueue - about to add: 21:10:11: Transaction task: WeatherData with 1 actions
21:10:11.512 [main] DEBUG n.f.g.s.b.BtLEQueue - about to add: 21:10:11: Transaction task: WeatherData with 1 actions
21:10:11.512 [main] DEBUG n.f.g.s.b.BtLEQueue - about to add: 21:10:11: Transaction task: WeatherData with 1 actions
21:10:11.512 [main] DEBUG n.f.g.s.d.p.PineTimeJFSupport - Wrote weather data
21:10:11.512 [Gadgetbridge GATT Dispatcher] DEBUG n.f.g.s.b.a.WriteAction - writing to characteristic: 00040001-78fc-48fe-8e23-433b3a1942d0: 0xbf 0x69 0x54 0x69 0x6d 0x65 0x73 0x74 0x61 0x6d 0x70 0x1a 0x65 0x1c 0x67 0x13 0x67 0x45 0x78 0x70 0x69 0x72 0x65 0x73 0x19 0x54 0x60 0x69 0x45 0x76 0x65 0x6e 0x74 0x54 0x79 0x70 0x65 0x07 0x68 0x4c 0x6f 0x63 0x61 0x74 0x69 0x6f 0x6e 0x6c 0x42 0x45 0x52 0x4c 0x49 0x4e 0x2d 0x54 0x45 0x47 0x45 0x4c 0x68 0x41 0x6c 0x74 0x69 0x74 0x75 0x64 0x65 0x00 0x68 0x4c 0x61 0x74 0x69 0x74 0x75 0x64 0x65 0x00 0x69 0x4c 0x6f 0x6e 0x67 0x69 0x74 0x75 0x64 0x65 0x00 0xff
21:10:11.615 [Binder:20849_9] DEBUG n.f.g.s.b.BtLEQueue - characteristic write: 00040001-78fc-48fe-8e23-433b3a1942d0 (success)
21:10:11.615 [Gadgetbridge GATT Dispatcher] DEBUG n.f.g.s.b.BtLEQueue - About to run action: 3. Oktober, 21:10: WriteAction on characteristic: 00040001-78fc-48fe-8e23-433b3a1942d0
21:10:11.616 [Gadgetbridge GATT Dispatcher] DEBUG n.f.g.s.b.a.WriteAction - writing to characteristic: 00040001-78fc-48fe-8e23-433b3a1942d0: 0xbf 0x69 0x54 0x69 0x6d 0x65 0x73 0x74 0x61 0x6d 0x70 0x1a 0x65 0x1c 0x67 0x13 0x67 0x45 0x78 0x70 0x69 0x72 0x65 0x73 0x19 0x54 0x60 0x69 0x45 0x76 0x65 0x6e 0x74 0x54 0x79 0x70 0x65 0x09 0x68 0x48 0x75 0x6d 0x69 0x64 0x69 0x74 0x79 0x18 0x60 0xff
21:10:11.714 [Binder:20849_9] DEBUG n.f.g.s.b.BtLEQueue - characteristic write: 00040001-78fc-48fe-8e23-433b3a1942d0 (success)
21:10:11.715 [Gadgetbridge GATT Dispatcher] DEBUG n.f.g.s.b.BtLEQueue - About to run action: 3. Oktober, 21:10: WriteAction on characteristic: 00040001-78fc-48fe-8e23-433b3a1942d0
21:10:11.716 [Gadgetbridge GATT Dispatcher] DEBUG n.f.g.s.b.a.WriteAction - writing to characteristic: 00040001-78fc-48fe-8e23-433b3a1942d0: 0xbf 0x69 0x54 0x69 0x6d 0x65 0x73 0x74 0x61 0x6d 0x70 0x1a 0x65 0x1c 0x67 0x13 0x67 0x45 0x78 0x70 0x69 0x72 0x65 0x73 0x19 0x54 0x60 0x69 0x45 0x76 0x65 0x6e 0x74 0x54 0x79 0x70 0x65 0x03 0x6b 0x54 0x65 0x6d 0x70 0x65 0x72 0x61 0x74 0x75 0x72 0x65 0x19 0x05 0xcd 0x68 0x44 0x65 0x77 0x50 0x6f 0x69 0x6e 0x74 0x39 0x7f 0xff 0xff
21:10:11.753 [main] DEBUG n.f.g.e.NotificationListener - Notification 0 posted: packageName=com.termux.api, priority=0, category=null
21:10:11.754 [main] INFO  n.f.g.e.NotificationListener - Allowing notification, application is whitelisted
21:10:11.755 [main] INFO  n.f.g.e.NotificationListener - Not forwarding notification, screen seems to be on and settings do not allow this
21:10:11.756 [main] INFO  n.f.g.e.NotificationListener - Ignoring notification: com.termux.api
21:10:11.816 [Binder:20849_9] DEBUG n.f.g.s.b.BtLEQueue - characteristic write: 00040001-78fc-48fe-8e23-433b3a1942d0 (success)
21:10:11.816 [Gadgetbridge GATT Dispatcher] DEBUG n.f.g.s.b.BtLEQueue - About to run action: 3. Oktober, 21:10: WriteAction on characteristic: 00040001-78fc-48fe-8e23-433b3a1942d0
21:10:11.817 [Gadgetbridge GATT Dispatcher] DEBUG n.f.g.s.b.a.WriteAction - writing to characteristic: 00040001-78fc-48fe-8e23-433b3a1942d0: 0xbf 0x69 0x54 0x69 0x6d 0x65 0x73 0x74 0x61 0x6d 0x70 0x1a 0x65 0x1c 0x67 0x13 0x67 0x45 0x78 0x70 0x69 0x72 0x65 0x73 0x1a 0x00 0x01 0x51 0x80 0x69 0x45 0x76 0x65 0x6e 0x74 0x54 0x79 0x70 0x65 0x03 0x6b 0x54 0x65 0x6d 0x70 0x65 0x72 0x61 0x74 0x75 0x72 0x65 0x19 0x05 0x9b 0x68 0x44 0x65 0x77 0x50 0x6f 0x69 0 [... cut ...]
21:10:12.405 [main] INFO  n.f.g.e.NotificationListener - Ignoring notification: com.termux.api
21:10:12.635 [main] DEBUG n.f.g.e.NotificationListener - Notification 0 posted: packageName=com.termux.api, priority=0, category=null
21:10:12.636 [main] INFO  n.f.g.e.NotificationListener - Allowing notification, application is whitelisted
21:10:12.638 [main] INFO  n.f.g.e.NotificationListener - Not forwarding notification, screen seems to be on and settings do not allow this
21:10:12.638 [main] INFO  n.f.g.e.NotificationListener - Ignoring notification: com.termux.api
</code></pre></details>

Avamander, 2023-10-06:
This is most likely the daily average. That will be disabled in the next GB update until it can be handled better.

closed 2023-10-06 by Avamander

joseph58tech, 2023-10-11:
Wait you can get weather data to pass through on GadgetBridge?

FintasticMan, 2023-10-11:
Yes, but it currently has a high chance of crashing the watch. See #1788.