InfiniTimeOrg/InfiniTime #1866: InfiniTime v1.12.0+ rejects writes from Android companion app Gadgetbridge

kimimaru4000, 2023-09-23:
### Verification

- [X] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

I cannot connect to my PineTime running InfiniTime 1.12.0 or above with Gadgetbridge.

### What should happen instead?

I should be able to connect to my PineTime running InfiniTime 1.12.0 or above with Gadgetbridge.

### Reproduction steps

1. I upgraded my PineTime to InfiniTime 1.12.0 (I tried 1.13.0 as well).
2. Try to connect to the PineTime with Gadgetbridge on Android.
3. Gadgetbridge stays stuck in a "Connecting..." state, seemingly unable to actually connect to the watch.

### More details?

This appears to be a regression, as connecting works with InfiniTime 1.11.0.

My PineTime is sealed, and I'm running Gadgetbridge on LineageOS 17.1, an Android 10 device.

I posted on the [Gadgetbridge issue tracker](https://codeberg.org/Freeyourgadget/Gadgetbridge/issues/3271) and left Gadgetbridge logs when attempting to connect to the PineTime. I'll repost them here:

```
00:27:29.045 [main] INFO  n.f.g.Logging - Gadgetbridge version: 0.75.1-bb17a0731
00:27:30.781 [main] DEBUG n.f.g.s.DeviceCommunicationService - Service startcommand: nodomain.freeyourgadget.gadgetbridge.devices.action.connect
00:27:30.787 [main] INFO  n.f.g.s.b.BtLEQueue - Attempting to connect to InfiniTime
00:27:30.789 [Gadgetbridge GATT Dispatcher] DEBUG n.f.g.s.b.BtLEQueue - Queue Dispatch Thread started.
00:27:30.797 [main] DEBUG n.f.g.s.b.BtLEQueue - new device connection state: CONNECTING
00:27:30.824 [main] INFO  n.f.g.s.DeviceCommunicationService - Setting broadcast receivers to: false
00:27:30.830 [main] INFO  n.f.g.s.DeviceCommunicationService - Setting broadcast receivers to: false
00:27:30.925 [main] DEBUG n.f.g.e.NotificationListener - Notification 1 posted: packageName=nodomain.freeyourgadget.gadgetbridge, priority=0, category=null
00:27:30.927 [main] INFO  n.f.g.e.NotificationListener - Not forwarding notification, screen seems to be on and settings do not allow this
00:27:30.927 [main] INFO  n.f.g.e.NotificationListener - Ignoring notification: nodomain.freeyourgadget.gadgetbridge
00:27:30.946 [main] DEBUG n.f.g.e.NotificationListener - Notification 1 posted: packageName=nodomain.freeyourgadget.gadgetbridge, priority=0, category=null
00:27:30.948 [main] INFO  n.f.g.e.NotificationListener - Not forwarding notification, screen seems to be on and settings do not allow this
00:27:30.949 [main] INFO  n.f.g.e.NotificationListener - Ignoring notification: nodomain.freeyourgadget.gadgetbridge
00:27:31.059 [Binder:31964_3] DEBUG n.f.g.s.b.BtLEQueue - connection state change, newState: 2 (success)
00:27:31.060 [Binder:31964_3] INFO  n.f.g.s.b.BtLEQueue - Connected to GATT server.
00:27:31.060 [Binder:31964_3] DEBUG n.f.g.s.b.BtLEQueue - new device connection state: CONNECTED
00:27:31.061 [Binder:31964_3] INFO  n.f.g.s.b.BtLEQueue - Attempting to start service discovery
00:27:31.062 [main] INFO  n.f.g.e.BluetoothConnectReceiver - connection attempt detected from EB:E5:12:75:F5:90(InfiniTime)
00:27:31.086 [main] INFO  n.f.g.s.DeviceCommunicationService - Setting broadcast receivers to: false
00:27:31.097 [Binder:31964_3] DEBUG n.f.g.s.d.p.PineTimeJFSupport - discovered unsupported service: Generic Access: 00001800-0000-1000-8000-00805f9b34fb
00:27:31.099 [Binder:31964_3] DEBUG n.f.g.s.d.p.PineTimeJFSupport - discovered unsupported service: Generic Attribute: 00001801-0000-1000-8000-00805f9b34fb
00:27:31.099 [Binder:31964_3] DEBUG n.f.g.s.d.p.PineTimeJFSupport - discovered supported service: Device Information: 0000180a-0000-1000-8000-00805f9b34fb
00:27:31.100 [Binder:31964_3] INFO  n.f.g.s.d.p.PineTimeJFSupport -     characteristic: Manufacturer Name String: 00002a29-0000-1000-8000-00805f9b34fb
00:27:31.100 [Binder:31964_3] INFO  n.f.g.s.d.p.PineTimeJFSupport -     characteristic: Model Number String: 00002a24-0000-1000-8000-00805f9b34fb
00:27:31.100 [Binder:31964_3] INFO  n.f.g.s.d.p.PineTimeJFSupport -     characteristic: Serial Number String: 00002a25-0000-1000-8000-00805f9b34fb
00:27:31.101 [Binder:31964_3] INFO  n.f.g.s.d.p.PineTimeJFSupport -     characteristic: Firmware Revision String: 00002a26-0000-1000-8000-00805f9b34fb
00:27:31.101 [Binder:31964_3] INFO  n.f.g.s.d.p.PineTimeJFSupport -     characteristic: Hardware Revision String: 00002a27-0000-1000-8000-00805f9b34fb
00:27:31.101 [Binder:31964_3] INFO  n.f.g.s.d.p.PineTimeJFSupport -     characteristic: Software Revision String: 00002a28-0000-1000-8000-00805f9b34fb
00:27:31.101 [Binder:31964_3] DEBUG n.f.g.s.d.p.PineTimeJFSupport - discovered supported service: Current Time Service: 00001805-0000-1000-8000-00805f9b34fb
00:27:31.101 [Binder:31964_3] INFO  n.f.g.s.d.p.PineTimeJFSupport -     characteristic: Current Time: 00002a2b-0000-1000-8000-00805f9b34fb
00:27:31.102 [Binder:31964_3] DEBUG n.f.g.s.d.p.PineTimeJFSupport - discovered supported service: Unknown Service: 00000000-78fc-48fe-8e23-433b3a1942d0
00:27:31.102 [Binder:31964_3] INFO  n.f.g.s.d.p.PineTimeJFSupport -     characteristic: Unknown Characteristic: 00000001-78fc-48fe-8e23-433b3a1942d0
00:27:31.102 [Binder:31964_3] INFO  n.f.g.s.d.p.PineTimeJFSupport -     characteristic: Unknown Characteristic: 00000002-78fc-48fe-8e23-433b3a1942d0
00:27:31.102 [Binder:31964_3] INFO  n.f.g.s.d.p.PineTimeJFSupport -     characteristic: Unknown Characteristic: 00000004-78fc-48fe-8e23-433b3a1942d0
00:27:31.103 [Binder:31964_3] INFO  n.f.g.s.d.p.PineTimeJFSupport -     characteristic: Unknown Characteristic: 00000003-78fc-48fe-8e23-433b3a1942d0
00:27:31.103 [Binder:31964_3] INFO  n.f.g.s.d.p.PineTimeJFSupport -     characteristic: Unknown Characteristic: 00000005-78fc-48fe-8e23-433b3a1942d0
00:27:31.103 [Binder:31964_3] INFO  n.f.g.s.d.p.PineTimeJFSupport -     characteristic: Unknown Characteristic: 00000006-78fc-48fe-8e23-433b3a1942d0
00:27:31.103 [Binder:31964_3] INFO  n.f.g.s.d.p.PineTimeJFSupport -     characteristic: Unknown Characteristic: 00000007-78fc-48fe-8e23-433b3a1942d0
00:27:31.104 [Binder:31964_3] INFO  n.f.g.s.d.p.PineTimeJFSupport -     characteristic: Unknown Characteristic: 00000007-78fc-48fe-8e23-433b3a1942d0
00:27:31.104 [Binder:31964_3] INFO  n.f.g.s.d.p.PineTimeJFSupport -     characteristic: Unknown Characteristic: 00000008-78fc-48fe-8e23-433b3a1942d0
00:27:31.104 [Binder:31964_3] INFO  n.f.g.s.d.p.PineTimeJFSupport -     characteristic: Unknown Characteristic: 00000009-78fc-48fe-8e23-433b3a1942d0
00:27:31.104 [Binder:31964_3] INFO  n.f.g.s.d.p.PineTimeJFSupport -     characteristic: Unknown Characteristic: 0000000a-78fc-48fe-8e23-433b3a1942d0
00:27:31.104 [Binder:31964_3] INFO  n.f.g.s.d.p.PineTimeJFSupport -     characteristic: Unknown Characteristic: 0000000b-78fc-48fe-8e23-433b3a1942d0
[... 81 lines cut ...]
00:27:31.731 [Binder:31964_3] INFO  n.f.g.s.AbstractDeviceSupport - Got event for VERSION_INFO: GBDeviceEventVersionInfo: fwVersion: 1.13.0; hwVersion: 1.0.0
00:27:31.731 [Gadgetbridge GATT Dispatcher] DEBUG n.f.g.s.b.BtLEQueue - About to run action: August 24, 00:27: NotifyAction on characteristic: 00000001-78fc-48fe-8e23-433b3a1942d0
00:27:31.733 [Gadgetbridge GATT Dispatcher] DEBUG n.f.g.s.b.a.NotifyAction - use NOTIFICATION
00:27:31.737 [main] INFO  n.f.g.s.DeviceCommunicationService - Setting broadcast receivers to: false
00:27:31.767 [main] DEBUG n.f.g.e.NotificationListener - Notification 1 posted: packageName=nodomain.freeyourgadget.gadgetbridge, priority=0, category=null
00:27:31.769 [main] INFO  n.f.g.e.NotificationListener - Not forwarding notification, screen seems to be on and settings do not allow this
00:27:31.769 [main] INFO  n.f.g.e.NotificationListener - Ignoring notification: nodomain.freeyourgadget.gadgetbridge
00:27:31.821 [Binder:31964_3] DEBUG n.f.g.s.b.BtLEQueue - descriptor write: 00002902-0000-1000-8000-00805f9b34fb (failed: 3)
00:27:31.822 [Binder:31964_3] DEBUG n.f.g.s.b.BtLEQueue - failed btle action, aborting transaction: 00000001-78fc-48fe-8e23-433b3a1942d0 (failed: 3)
00:27:31.841 [main] DEBUG n.f.g.e.NotificationListener - Notification 1 posted: packageName=nodomain.freeyourgadget.gadgetbridge, priority=0, category=null
00:27:31.844 [main] INFO  n.f.g.e.NotificationListener - Not forwarding notification, screen seems to be on and settings do not allow this
00:27:31.844 [main] INFO  n.f.g.e.NotificationListener - Ignoring notification: nodomain.freeyourgadget.gadgetbridge
```

The conclusion was that the PineTime could be preventing some writes. I understand this is not the Gadgetbridge repository, so I'm looking for someone here has more knowledge on what has changed in the Bluetooth/connection stack between 1.11.0 and 1.12.0.

I'm happy to provide additional information.

### Version

v1.12.0+

### Companion app

Gadgetbridge v0.76.0 (c912ea037)

mark9064, 2023-09-23:
Have you tried removing the device from Gadgetbridge and re-adding it? You can export the device data from Gadgetbridge before you remove it and import it after you add it again to preserve your step history etc.

kimimaru4000, 2023-09-23:
Hi @mark9064, yes I tried that right now, but it cannot find the device running the new firmware (v1.13.0) at all unfortunately.

mark9064, 2023-09-23:
Try going into your android bluetooth settings and making sure all pairings are removed. Then search in Gadgetbridge as before

kimimaru4000, 2023-09-23:
Done, but still no dice as there weren't any pairings leftover. Thanks for your help thus far!

mark9064, 2023-09-23:
Hmm, maybe try restarting your phone then? AFAIK there haven't been any bluetooth stack changes, and if it's possible to connect on another device then bluetooth appears to be working

ldsvrn, 2023-09-23:
I've had a similar issue last week, pressing the side button until the watch restarted and repairing it to Gadgetbridge did the trick for me.

kimimaru4000, 2023-09-24:
Thanks @ldsvrn that ended up working for me!

closed 2023-09-24 by kimimaru4000

Zykino, 2023-09-30:
I’m in the same situation, just received 2 watch on 1.11. After the update to 1.13 (or 1.12) they do not connect back to GadgetBridge. I just try @ldsvrn’s solution… but it "just" revert to 1.11, just like going to the firmware menu and pressing "revert".

Did you press "Validate" in the firmware menu first? Is it really safe to do since that means possibly accepting a version that does not work over bluetooth?

kimimaru4000, 2023-10-01:
@Zykino Yes, I had validated the firmware as I was able to connect v1.12+ with [WatchMate](https://github.com/azymohliad/watchmate) over bluetooth before doing the reset steps to have it connect to Gadgetbridge.

Zykino, 2023-10-01:
Ok. So I think I had to do everything for it to work in the end:
1. Let the watch boot, try to send it an update that did not work (I think it did not catch the Bluetooth at that time since I don’t remember seeing its icon).
    1. Tell the phone and GadgetBridge to forget the watch… which was useless at that point.
1. Then reboot the watch (at least) once.
1. Update again, see that the watch think it is connected while the phone don’t think so.
    1. I redid this a few times and also fond out that WatchMate worked while GadgetBridge did not…
1. Then in discord someone told me that the "number pass" that is used to connect to the phone act as a password and it needs to be redone since the OS changed. So I re-asked the phone and GadgetBridge to forget the watch and it finally worked… mostly.

I said mostly because the 2nd watch had trouble a couple of times to transfer the update once the first one was updated. (But it looked to be sending/Bluetooth errors.)