InfiniTimeOrg/InfiniTime #1867: RAM overflow

JosuGZ, 2023-09-28:
### Verification

- [X] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

I'm trying to build the project but I get a RAM overflow on the linking step.

### Reproduction steps

I'm using the dev container. Inside, I run the following commands (after installing node + some libs wich is another topic):

```bash
mkdir build && cd build && cmake \
  -DARM_NONE_EABI_TOOLCHAIN_PATH=/opt/gcc-arm-none-eabi-9-2020-q2-update \
  -DNRF5_SDK_PATH=/opt/nRF5_SDK_15.3.0_59ac345 \
  -DCMAKE_BUILD_TYPE=Release .. \
  -DTARGET_DEVICE=PINETIME

make -j4 pinetime-app
```

### More details?

```
[100%] Linking CXX executable pinetime-app-1.13.0.out
Memory region         Used Size  Region Size  %age Used
           FLASH:      377984 B     474632 B     79.64%
     SPARE_SPACE:          0 GB        12 KB      0.00%
             RAM:       64784 B        64 KB     98.85%
/opt/gcc-arm-none-eabi-9-2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: region RAM overflowed with stack
```

### Version

Latest commit on main.

mark9064, 2023-09-28:
I believe it is because your GCC is too old, you need 10.3-2021.10. See https://github.com/InfiniTimeOrg/InfiniTime/blob/main/doc/buildAndProgram.md
I can't speak for the dev container because I don't use it, but I suspect the compiler version is the problem

JosuGZ, 2023-09-28:
I'm seeing that there is a docker folder which looks similar to the .devcontainer folder. Perhaps the devcontainer should reverence the docker folder as it seems outdated.

The latest docker container also fails though :'( (`Error: @swc/core threw an error when attempting to validate swc compiler options.`).

I can't see older images without signing to docker hub apparently.

FintasticMan, 2023-10-07:
See #1587 for the devcontainer.

abukharmeh, 2023-10-22:
I have debug builds ram overflowing on `gcc-arm-none-eabi-10.3-2021.10` only on Debug builds. Removing the logging functionality seems to make it work !

JF002, 2023-12-10:
InfiniTime statically allocates most of the RAM memory and we check that the project builds with the version of the compiler we mention in the documentation (10.3-2021.10 as of now). 

Building with another version of the compiler might change the way the memory is allocated and could indeed fail. 
Also, building in debug mode with the logging feature enable will definitely allocate more RAM memory which will result in memory overflow at link time. 
If you want to temporarily build in DEBUG and/or enable logging, you can reduce the size of the main memory heap by editing  [this constant](https://github.com/InfiniTimeOrg/InfiniTime/blob/main/src/FreeRTOSConfig.h#L65). I regularly set `configTOTAL_HEAP_SIZE` to `1024*35` when I enable the logging functionality. 
Please not that this reduce the total amount of memory available to InfiniTime and the apps so this could trigger some issues in memory hungry apps.

closed 2023-12-10 by JF002