InfiniTimeOrg/InfiniTime #1868: infinitime/infinitime-build:latest fails to build

JosuGZ, 2023-09-28:
### Verification

- [X] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

Building with the docker image fails

### Reproduction steps

After cloning main:

`docker run --rm -it -v ${PWD}:/sources --user $(id -u):$(id -g) infinitime/infinitime-build:v2 /opt/build.sh pinetime-app`

I get the following error:

```
Traceback (most recent call last):
  File "/sources/src/resources/generate-img.py", line 56, in <module>
    main()
  File "/sources/src/resources/generate-img.py", line 51, in main
    subprocess.check_call(line)
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/sources/node_modules/.bin/lv_img_conv', '/sources/src/resources/images/navigation1.png', '--force', '--output-file', 'navigation1.bin', '--color-format', 'CF_INDEXED_1_BIT', '--output-format', 'bin', '--binary-format', 'ARGB8565_RBSWAP']' returned non-zero exit status 243.
gmake[3]: *** [src/resources/CMakeFiles/GenerateResources.dir/build.make:73: src/resources/CMakeFiles/GenerateResources] Error 1
gmake[2]: *** [CMakeFiles/Makefile2:686: src/resources/CMakeFiles/GenerateResources.dir/all] Error 2
gmake[2]: *** Waiting for unfinished jobs...
```

### Version

Latest commit on main.

FintasticMan, 2023-09-29:
Duplicate of #1859. Will be fixed by either #1863 or #1764.

closed 2023-09-29 by FintasticMan

JosuGZ, 2023-09-29:
The package-lock.json should help a lot preventing problems when new images are created.