InfiniTimeOrg/InfiniTime #1871: Unable to build with DBUILD_TARGET MOY-TFK5

CutieCat6778, 2023-10-03:
### Verification

- [X] I searched for similar issues (including closed issues) and found none was relevant.

### Introduce the issue

I'm trying to build InfiniTime with Docker and configure the build.sh script to add -DBUILD-TARGET=MOY-TFK5

But at the end, I recieve this, while trying to run the script

```console
BUILD CONFIGURATION
-------------------
    * Mode : Release
    * Version : 1.13.0
    * Toolchain : /opt/gcc-arm-none-eabi-10.3-2021.10
    * GitRef(S) : 3a7dfdba
    * NRF52 SDK : /opt/nRF5_SDK_15.3.0_59ac345
    * Target device : MOY-TKF5
    * Build DFU (using adafruit-nrfutil) : Enabled
    * Build resources : Enabled
CMake Error at src/CMakeLists.txt:827 (message):
  Invalid TARGET_DEVICE
  ````

CutieCat6778, 2023-10-03:
It finally started to build, but I got this error message:

```console
gmake[2]: *** [src/CMakeFiles/pinetime-recovery.dir/build.make:76: src/CMakeFiles/pinetime-recovery.dir/stdlib.c.o] Error 1
cc1plus: all warnings being treated as errors
cc1plus: all warnings being treated as errors
gmake[2]: *** [src/CMakeFiles/pinetime-recovery.dir/build.make:118: src/CMakeFiles/pinetime-recovery.dir/logging/NrfLogger.cpp.o] Error 1
gmake[2]: *** [src/CMakeFiles/pinetime-recovery.dir/build.make:90: src/CMakeFiles/pinetime-recovery.dir/FreeRTOS/heap_4_infinitime.c.o] Error 1
gmake[2]: *** [src/CMakeFiles/pinetime-recovery.dir/build.make:174: src/CMakeFiles/pinetime-recovery.dir/drivers/SpiNorFlash.cpp.o] Error 1
gmake[2]: *** [src/CMakeFiles/pinetime-recovery.dir/build.make:202: src/CMakeFiles/pinetime-recovery.dir/drivers/Spi.cpp.o] Error 1
cc1plus: all warnings being treated as errors
cc1plus: all warnings being treated as errors
gmake[2]: *** [src/CMakeFiles/pinetime-recovery.dir/build.make:160: src/CMakeFiles/pinetime-recovery.dir/drivers/St7789.cpp.o] Error 1
gmake[2]: *** [src/CMakeFiles/pinetime-recovery.dir/build.make:216: src/CMakeFiles/pinetime-recovery.dir/drivers/Watchdog.cpp.o] Error 1
cc1plus: all warnings being treated as errors
gmake[2]: *** [src/CMakeFiles/pinetime-recovery.dir/build.make:132: src/CMakeFiles/pinetime-recovery.dir/displayapp/DisplayAppRecovery.cpp.o] Error 1
gmake[2]: *** [src/CMakeFiles/pinetime-recovery.dir/build.make:188: src/CMakeFiles/pinetime-recovery.dir/drivers/SpiMaster.cpp.o] Error 1
gmake[2]: *** [src/CMakeFiles/pinetime-recovery.dir/build.make:230: src/CMakeFiles/pinetime-recovery.dir/drivers/InternalFlash.cpp.o] Error 1
cc1plus: all warnings being treated as errors
gmake[2]: *** [src/CMakeFiles/pinetime-recovery.dir/build.make:146: src/CMakeFiles/pinetime-recovery.dir/main.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:358: src/CMakeFiles/pinetime-recovery.dir/all] Error 2
gmake: *** [Makefile:91: all] Error 2
```

Avamander, 2023-10-03:
The actual error message, however, seems to be missing from the output you pasted. 

If you can, please post the full output.

CutieCat6778, 2023-10-03:
@Avamander yes, this is the full output: https://pastebin.com/3QtdSgKQ

CutieCat6778, 2023-10-03:
Tried to run with Ninja Generator and the output is this: https://pastebin.com/21SHe2x4

CutieCat6778, 2023-10-03:
Found a solution with the help of @FintasticMan by removing the `804` and `805` from `src/CMakeLists.txt`.

closed 2023-10-03 by CutieCat6778