InfiniTimeOrg/InfiniTime #1878: RFC: Timeline event packing

Avamander, 2023-10-06:
Problem:
It's somewhat difficult to handle long-term and short-term events overlapping in the timeline in user-facing situations.

Proposal:
Some of this confusion can be removed by implementing predictable conflict resolution. 

We can adjust the `timestamp` value of the longer event to start at the end of shorter/previous element's `timestamp + expiration` and shorten expiration of the longer event by the amount shortened.

This packing shouldn't be done for equally-long events, neither should a shorter event be shortened.

Before:
```
|     |     |     |     |     |     |     |     |     |
|────────────────────────────────────────────────────────────►
  ┌───────────┐
  │Temperature│
  └───────────┘
     ┌────────────────────────────────────────────────┐
     │ Temperature (long-term)                        │
     └────────────────────────────────────────────────┘
```
After:
```
|     |     |     |     |     |     |     |     |     |
|────────────────────────────────────────────────────────────►
  ┌───────────┐┌──────────────────────────────────────┐
  │Temperature││ Temperature (long-term)              │
  └───────────┘└──────────────────────────────────────┘
```

Are there any potential use-cases where this might not work out as intended?

minacode, 2023-10-07:
> It's somewhat difficult to handle long-term and short-term events overlapping in the timeline in user-facing situations.

Can you explain more, what exactly the problem is?

JF002, 2023-12-23:
Thanks for your work @Avamander. However, I decided to rework the weather service to fix the issues more easily, so I'll close this one in favor of #1924.

closed 2023-12-23 by JF002