InfiniTimeOrg/InfiniTime #1888: Recieving many notifications in a short timespan causes the watch to get soft bricked

ThePinkUnicorn6, 2023-10-11:
### Verification

- [X] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

When the watch recieves a lot of notifications within a second, the watch reboots into the recovery mode and will not boot up to the watchface again.

### What should happen instead?

The notifications should be displayed and then it should go back to the watch face.

### Reproduction steps

In the Gadgetbridge app on android, go to the debug menu that is in the side bar. Then press the "Send" button as fast as you can for a few seconds. The watch will display the first few notifications but then reboot into the recovery mode. Restarting the watch by holding down the button for a few seconds will reboot it back into recovery mode. By reconecting the watch to gadget bridge and re-uploading the firmare the device was able to boot to the watchface again.

### More details?

The bug could be triggered by a badly written app that sends a lot of notifications (eg. some music apps). 

### Version

v1.13.0

### Companion app

Gadgetbridge 0.76.1

ThePinkUnicorn6, 2023-10-11:
I just tried to trigger the bug again but interestingly it did not cause it to boot into the recovery mode when it restarted.

minacode, 2023-10-14:
This is my guess what happened:
1. nimble tries to allocate more memory than it has available
2. it crashes
3. you didn't verify your firmware before
4. the watch boots into recovery

ThePinkUnicorn6, 2023-10-14:
I the firmware was validated before doing this and I had restarted the watch multiple times before this happened. I cannot get the watch to get stuck in recovery again though so it must require specific timing, or the watch to be using over a certain amount of memory, or something like that.

minacode, 2023-10-14:
Is it possible to overwrite or corrupt the current firmware with a buffer overflow? I don't know where the firmware is stored.

erobertsme, 2023-10-25:
Similar to this my Pinetime will reboot with rapid notifications.

My phone sometimes displays notifications from my work email on Google Workspace all at once which causes the Pinetime to get spammed with notifications and it reboots.

Oddly, it got hit with a ton of notifications today from this but only showed a few and did not reboot.

JF002, 2023-11-12:
I can reproduce the crash with the debug screen in Gadgetbridge.

The issue is caused by SystemTask and DisplayApp sending each other messages (via their respective message queues) faster than they can process them. At some points, the messages queues are full and both tasks are in a deadlock situation.

I submitted a fix in #1911.

Regarding the soft-brick part of this post, it was probably caused by the firmware that was not properly validated (in the settings menu) when the crash occurred. While maybe not impossible, there's very few chances the firmware could corrupt itself and force a reboot in recovery mode.

PR #1911 (2023-11-19): Fix deadlock when a lot of notifications are received in a very short time span
When a notification is received, SystemTask sends messages to DisplayApp, which loads the Notifications apps that sends a few messages to SystemApp. When notification are received too quickly, DisplayApp and SystemTask cannot process those messages fast enough (probably because of the time it takes to refresh the display) and the message queues fill up. When they are full, the current implementation just waits until there's room available to store the event. In this case, since both tasks exchange messages, they en [... cut]
closed 2023-11-19 by JF002