InfiniTimeOrg/InfiniTime #1895: AlertNotification Service/Client are not compliant to the spec

JF002, 2023-10-23:
### Verification

- [X] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

AlertNotification Service/Client use a 3 bytes header, the spec defines a 2 bytes header

### What should happen instead?

AlertNotification Service/Client should comply to the BLE spec.

### Reproduction steps

N/A

### More details?

This issue was reported by the [Amazfish community](https://github.com/piggz/harbour-amazfish/pull/286) : InfiniTime expects the header of all notifications is a 3 bytes header ([here](https://github.com/InfiniTimeOrg/InfiniTime/blob/main/src/components/ble/AlertNotificationService.cpp#L50) and [here](https://github.com/InfiniTimeOrg/InfiniTime/blob/main/src/components/ble/AlertNotificationClient.cpp#L147)).

However, the GATT spec [specifies a header of 2 bytes](https://github.com/oesmith/gatt-xml/blob/master/org.bluetooth.characteristic.new_alert.xml).

Now, even if InfiniTime is not 100% compliant, all companion apps have currently implemented this 3 bytes header, and changing this might break the compatibility with those companion apps so I'm not sure if we should fix this. 
If we do want to change the header size, we'll have to communicate with companion app developers to ensure smooth transition.

It might be possible to support both 2 and 3 bytes header by checking the value of the 3rd byte of the buffer : it can be ignored if it's equal to 0x00 or lower than 0x20.

The [documentation](https://github.com/InfiniTimeOrg/InfiniTime/blob/main/doc/ble.md#new-alert) should however be completed to mention that this is not completely compliant to the spec of the AlertNotificationService.

### Version

<=1.13.0

### Companion app

All

KaffeinatedKat, 2023-10-24:
>It might be possible to support both 2 and 3 bytes header by checking the value of the 3rd byte of the buffer : it can be ignored if it's equal to 0x00 or lower than 0x20.

This might be a good temporary solution. Support both while transitioning all the companion apps, then the support for the 3 byte header can be removed once all the apps have moved over to comply with the new header size

jmlich, 2023-10-27:
For Amazfish is it oneline to change.

minacode, 2023-11-05:
I vote to change it, because not being standard compliant only gets worse with time, right?

Nobody2303, 2023-11-07:
> I vote to change it, because not being standard compliant only gets worse with time, right?

agreed 🤝