InfiniTimeOrg/InfiniTime #1905: Can't build the docker image on Win 10

Nobody2303, 2023-11-07:
### Verification

- [X] I searched for similar issues (including closed issues) and found none was relevant.

### Introduce the issue

I just set up Docker v4.25.0
I cloned the infinitime repo and also update the submodules.

Upon trying to build the docker image, I ran into the following issue, which I can't really resolve as it's the first time for me to work with docker.

```$ winpty docker build -t infinitime-build ./docker

[+] Building 1.7s (13/15)                                                                                                                   docker:default
 => [internal] load .dockerignore                                                                                                                     0.0s
 => => transferring context: 2B                                                                                                                       0.0s
 => [internal] load build definition from Dockerfile                                                                                                  0.0s
 => => transferring dockerfile: 1.36kB                                                                                                                0.0s
 => [internal] load metadata for docker.io/library/ubuntu:22.04                                                                                       1.1s
 => [auth] library/ubuntu:pull token for registry-1.docker.io                                                                                         0.0s
 => [ 1/10] FROM docker.io/library/ubuntu:22.04@sha256:2b7412e6465c3c7fc5bb21d3e6f1917c167358449fecac8176c6e496e5c1f05f                               0.0s
 => [internal] load build context                                                                                                                     0.0s
 => => transferring context: 30B                                                                                                                      0.0s
 => CACHED [ 2/10] RUN apt-get update -qq     && apt-get install -y       bash       build-essential       cmake       git       make       python3   0.0s
 => CACHED [ 3/10] RUN pip3 install adafruit-nrfutil                                                                                                  0.0s
 => CACHED [ 4/10] RUN pip3 install -Iv cryptography==3.3                                                                                             0.0s
 => CACHED [ 5/10] RUN pip3 install cbor                                                                                                              0.0s
 => CACHED [ 6/10] RUN npm i lv_font_conv@1.5.2 -g                                                                                                    0.0s
 => CACHED [ 7/10] COPY build.sh /opt/                                                                                                                0.0s
 => ERROR [ 8/10] RUN bash -c "source /opt/build.sh; GetGcc;"                                                                                         0.4s
------
 > [ 8/10] RUN bash -c "source /opt/build.sh; GetGcc;":
: invalid option.sh: line 5: set: -
0.348 set: usage: set [-abefhkmnptuvxBCHP] [-o option-name] [--] [arg ...]
: invalid option.sh: line 6: set: -
0.348 set: usage: set [-abefhkmnptuvxBCHP] [-o option-name] [--] [arg ...]
0.348 /opt/build.sh: line 7: $'\r': command not found
0.348 /opt/build.sh: line 13: $'\r': command not found
0.349 /opt/build.sh: line 17: $'\r': command not found
0.350 /opt/build.sh: line 21: $'\r': command not found
0.352 /opt/build.sh: line 24: $'\r': command not found
0.352 /opt/build.sh: line 26: $'\r': command not found
0.353 /opt/build.sh: line 27: syntax error near unexpected token `$'{\r''
'.353 /opt/build.sh: line 27: `main() {
0.353 bash: line 1: GetGcc: command not found
------
Dockerfile:48
--------------------
  46 |     # Lets get each in a separate docker layer for better downloads
  47 |     # GCC
  48 | >>> RUN bash -c "source /opt/build.sh; GetGcc;"
  49 |     # NrfSdk
  50 |     RUN bash -c "source /opt/build.sh; GetNrfSdk;"
--------------------
ERROR: failed to solve: process "/bin/sh -c bash -c \"source /opt/build.sh; GetGcc;\"" did not complete successfully: exit code: 127
```

If anyone could give me a hint what might not work I'd be very happy.

### Version

v1.11.0

Nobody2303, 2023-11-09:
So I did a little Research and knowledge building about docker in it self on my own. 
I was able to alter the Dockerfile to get a plain image which i can run with sleep infinity and debug contents with docker exec.

What I found out is for some reason the RUN commands seemingly are not performed as a specific user so /opt/build.sh will not be executable.

Additionally I had to convert all CarriageReturn LineFeeds to LineFeeds only, most certainly due to the way how I cloned the git repo to my windows machine... should also considering re-cloning the whole repo with proper line feed alteration settings, like pull as is

By using the following lines I can make sure I see what is in the /opt folder during image build time and build.sh will actually execute

'''
# Lets get each in a separate docker layer for better downloads
# GCC
RUN su root && export PATH=$PATH:/opt && ls -l /opt && build.sh GetGcc
'''

but unfortunately this will start a complete build process not just installing GCC so the process is asking for CMakeLists.txt to be inside /sources folder inside the container during docker image build time, which makes no sense to me.

Could someone please elaborate how this is intended to work?

- Are we supposed to build the image and then transfer the repo content into the docker container to build?
- Do we issue a git command from within the container to get the sources in the compilation environment?
- Should we mount /sources to be the repo folder on my C: drive for the process to work?

Anyhow currently image build will only be a success if I don't try to execute build.sh at all

Nobody2303, 2023-11-11:
So after some further research and tweaking I got the container to work properly.
I had to add the following line at the 3rd or 4th line 
```USER root ``` to explicitly run the commands as root otherwise they would not succeed.
The docker run command from the docentation also needed to be altered to my specific folder structure as well as --user root to actually run the build process properly. 
Well at least to the point where nimble lib is build then the build exits with error 2 at 35% of total build process... also some font file it seems to be not created as well due to an error...

Nobody2303, 2023-11-15:
So Finally I was able to build the stock Firmware from the Repository and have a relyable start point for further development.

```
Memory region         Used Size  Region Size  %age Used
           FLASH:      377492 B     474632 B     79.53%
     SPARE_SPACE:          0 GB        12 KB      0.00%
             RAM:       64368 B        64 KB     98.22%
post build steps for pinetime-mcuboot-app-1.13.0
   text    data     bss     dec     hex filename
 377492     940   63420  441852   6bdfc pinetime-mcuboot-app-1.13.0.out
post build (DFU) steps for pinetime-mcuboot-app-1.13.0
Zip created at pinetime-mcuboot-app-dfu-1.13.0.zip
[100%] Built target pinetime-mcuboot-app
+ BUILD_RESULT=0
+ return 0
+ BUILD_RESULT=0
```

The remaining issue was now related to Line End conversion caused by the "auto" option in the [.gitattributes](https://github.com/InfiniTimeOrg/InfiniTime/commit/f6927548419542ce1540b6c1419b1c854fd17c8c) definition text = auto which overrides the [.gitconfig](https://github.com/InfiniTimeOrg/InfiniTime/commit/04fd4a7d03fe88a44807ca62b3697dfe5864b56c) autocrlf = input on initial cloning of the repository. 

autocrlf will only work on commit afaik when Line endings have been converted due to text = auto.

should I commit my changes to get docker working from the get go for review? @JF002 @NeroBurner @Riksu9000 @minacode

FintasticMan, 2023-11-28:
Yes, I would add that to the docs in a PR.

JF002, 2023-12-10:
@Nobody2303 Thanks for documenting all your findings about building the docker image on Win 10!
Do you have any idea if it would be possible to edit the docker file and scripts such that it would build seamlessly on Linux AND Windows? 

If I understand correctly, the image is built and ran as user  `root`, which, to the point of view of a Linux user, does not seem right (this will create build artifacts in the user directory that are owned by root instead of the actual user.

Nobody2303, 2023-12-13:
Sry for the late reply. Christmas preparations are keeping me from pursuing this topic. 

Currently I have no concrete Idea but I think it should be possible to select a fitting dockerfile and infinitime build command from within an OS detection mechanism...

PR #2181 (2025-06-16): Set LF-only (\n) line endings for .sh files to fix #1905
Windows machines may fail to build the docker image with cryptic bash/`set -` error messages if shell scripts are not checked out with unix-style line endings.

Should fix #1905