InfiniTimeOrg/InfiniTime #1929: Vibration too weak. Needs feature to adjust it.

5paceghost, 2023-12-18:
### Verification

- [X] I searched for similar issues (including closed issues) and found none was relevant.

### Introduce the issue

The vibration is too weak. This is an issue especially when a timer is set: the weak vibration is easily ignored and it also just vibrates once in this use case.

### Preferred solution

Please add the feature to set the vibration to low, medium and high. If this is not possible then it would suffice to set the length of the vibration or the frequency, i.e.: vibrate once, twice...  etc. 

### Version

v.1.13

kyeastmood, 2023-12-28:
I have the same problem, I often miss the small vibration in the timer. It would be great if we could specify vibration count or even make it indefinite like in the alarm.

kyeastmood, 2023-12-28:
Related issue: #391 
Having the countdown timer continue to work after it reached 0:00 would also be nice.

TheRealDestroyer, 2024-01-08:
i also think we should be able to adjust how strong it is for notifications and possibly be able to change the vibration pattern (e.g. you tap the screen/push the button to create a pattern that the watch uses for notifications)

vladopajic, 2024-01-18:
+1

vibration is too weak. 

on other watches i used vibration lasted ~2x longer and had 3 impulses in total. vibration pattern was {vibrate, pause, vibrate, pause, vibrate} where vibrate and pause intervals have the same length (which is ~2x longer that currently on pinewatch).

please note that vibration should be harder, because when people use it in sport activities it is vary hard to notice current vibration when you are active and moving. 

@JF002

Cottigniem, 2024-01-21:
+1
I just started using my PineTime updated to the latest version and I was searching for a setting to increase the vibration of the notifications. It would be great if this feature could be added :-).

psychedup, 2024-01-22:
+1
Any way to make it more noticeable (especially for the timer) would be appreciated.

Rathmox, 2024-01-26:
+1
It should also vibrate repeatedly until you click on the watch

hellfire103, 2024-07-14:
This issue is the only reason I opened GitHub today. Make it happen, you spectacular devs!

j0chn1, 2024-09-21:
> +1 It should also vibrate repeatedly until you click on the watch

This should be a settings option.
Generally od like to see the setting options for
Duration, short medium long
Intensity? Low medium and high
Pattern, once, sos, and alternating
And each pattern should be set for all or individually for notificationa, alarm, and timer.

Xorger, 2024-10-04:
+1
This is a really good idea, as it's the one thing keeping me from using the timer app.

I don't know if it's possible to adjust the vibration strength (possible hardware limitations) but i do think it would be great if we could choose the vibration duration (btw, this is ideally for all vibrations and not just the timer) and weather it should continue indefinitely. Maybe even the pattern! I hope JF sees this (not pinging because he already got pinged once)!

codeguylevi, 2025-02-16:
I really agree to this. Sometimes I don't notice a timer going off.