InfiniTimeOrg/InfiniTime #1932: Heart rate monitoring in the background seems to measure 0 BPM after turning it on, exiting the app and the watch going to sleep

IPlayZed, 2023-12-19:
### Verification

- [X] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

Heart rate monitoring in the background seems to measure 0 BPM after turning it on, exiting the app and the watch going to sleep

### What should happen instead?

Heart rate is shown on main watchface accurately whenever I look at it.

### Reproduction steps

1. Select heart rate monitoring application.
2. Start monitoring.
3. Wait for heart rate.
4. Once heart rate is displayed go to main watch face. Now it contains a heart rate value.
5. Let the screen to shut down.
6. Check back after some minutes to the main watch face, BPM is not nothing but zero instead.

### More details?

I think this is related to #183, but the problem still exists years later...

### Version

1.13.0 (Short ref 5f3acfad)

### Companion app

Watchmate 0.5.1

FintasticMan, 2023-12-19:
This is indeed a duplicate of #183, and will be fixed by #1718.

closed 2023-12-19 by FintasticMan

IPlayZed, 2023-12-19:
> This is indeed a duplicate of #183, and will be fixed by #1718. You can use the DFU in the CI for that PR to test it if you don't want to compile it yourself.

Thanks for the confirmation! I do not want to test it as my watch is sealed, but it is good to know it is in the works.

FintasticMan, 2023-12-19:
It is practically completely safe to flash InfiniTime DFU files, due to our robust bootloader and recovery firmware. If the watch resets without the running firmware being validated, it will revert to the previous firmware.

You can also boot into the recovery firmware by holding the button during boot until the pinecone turns red. That will allow you to connect over bluetooth and flash a DFU.

Of course it's completely understandable if you still wouldn't want to.

SnejPro, 2023-12-29:
> It is practically completely safe to flash InfiniTime DFU files, due to our robust bootloader and recovery firmware. If the watch resets without the running firmware being validated, it will revert [...]

Hi,
I'd like to test this firmware because I have some issues with heart rate monitoring and I miss the background monitoring feature.

How can I get the DFU? I cannot find the file here.

Greetings
Jens

FintasticMan, 2023-12-30:
I forgot to allow the workflow to run, so there wasn't a build available. Now there should be if you scroll down and click details next to CI / build-firmware, then summary, then the DFU file.

SnejPro, 2023-12-30:
> I forgot to allow the workflow to run, so there wasn't a build available. Now there should be if you scroll down and click details next to CI / build-firmware, then summary, then the DFU file.

Thx, now it works 👍