InfiniTimeOrg/InfiniTime #1951: Flashing resources mostly unavailable

zyphlar, 2024-01-06:
### Verification

- [X] I searched for similar issues (including closed issues) and found none was relevant.

### Introduce the issue

`itd` (https://gitea.elara.ws/Elara6331/itd) is currently not working too well due to [this issue](https://gitea.elara.ws/Elara6331/itd/issues/65) and GadgetBridge [doesn't yet support uploading resources](https://codeberg.org/Freeyourgadget/Gadgetbridge/issues/2919) so without digging into another project and stack I'm unable to upload new resources to my sealed PineTime.

### Preferred solution

It seems like those issues are *almost* fixed, if someone could give those projects a nudge it'd help greatly with the move to i.e. external resource imagery.

### Version

main

sdeneefe, 2024-08-23:
Hi, I am also stuck on this issue.  How is anyone currently updating their resources?  I am an android user, and I know gadgetbridge doesn't support it (yet), though someone was working on it.  I have both mac and linux machines, if that helps (but not a sailfish OS machine).  Surely someone has a way... or does no one ever update these?

zyphlar, 2024-08-23:
@sdeneefe IIRC WatchMate worked best last time I tried

mark9064, 2024-08-23:
AFAIK itd has been fixed as well and now works properly again for resources

SteveAmor, 2024-08-24:
Have you tried https://infinitimeexplorer.netlify.app/ 
IIRC you need to unzip the resources.

rnwgnr, 2025-01-11:
> AFAIK itd has been fixed as well and now works properly again for resources

It is fixed in `main`, but no "release" has been tagged since. So depending on the distro you are running you need to compile from `main` by yourself.