InfiniTimeOrg/InfiniTime #1954: Can't update to 1.14.0

w1ll1am23, 2024-01-06:
### Verification

- [X] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

Gadgetbridge update fails at 14%

### What should happen instead?

Update should complete

### Reproduction steps

Using the latest version of Gadgetbridge (0.77.0) and the pinetime-mcuboot-app-dfu-1.14.0.zip Gadgetbridge starts the upload and makes it to 14% 165k the watch is stuck on 0% and then crashes. Gadgetbridge indicates upload failed.

### Version

1.13.0

### Companion app

Gadgetbridge 0.77.0

wdh25, 2024-01-06:
Had a similar issue yesterday updating. Realized it's Android 14 on my phone. Connected to Gadgetbridge on a phone running Android 12 updated first try.

FintasticMan, 2024-01-06:
Yeah, unfortunately it's a bug in GadgetBridge. JF has made a PR that fixes the issue, which was merged recently. It will make its way into the next release of GadgetBridge. For now, the best way would either be to use a different device, or use [nRF Connect v4.24.3](https://github.com/NordicSemiconductor/Android-nRF-Connect/releases/tag/v4.24.3).

closed 2024-01-06 by FintasticMan

w1ll1am23, 2024-01-06:
Pulled out an old device with Android 11 and was able to flash. Thanks.