InfiniTimeOrg/InfiniTime #1955: watch stuck on Infinitime screen

djpeter84, 2024-01-07:
### Verification

- [X] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

watch does not update firmware

### What should happen instead?

update firemware and reboot

### Reproduction steps

trying to update to 1.14.0 https://github.com/InfiniTimeOrg/InfiniTime/releases/download/1.14.0/pinetime-mcuboot-app-dfu-1.14.0.zip , watch screen is InfiniTime but when i try to install, watch Infinitime turns green, upload hits 1% then i get Upload has failed. 

### More details?

this is similar to https://github.com/InfiniTimeOrg/InfiniTime/issues/1337  and https://github.com/InfiniTimeOrg/InfiniTime/discussions/1674 

### Version

1.14.0

### Companion app

gadgetbridge

FintasticMan, 2024-01-07:
Are you on Android 14? There is currently a bug with GadgetBridge on Android 14 that causes firmware updates to not work. It'll be fixed in the next GadgetBridge release. There's a section in the 1.14 release notes about what you can do in the meantime.

djpeter84, 2024-01-07:
> Are you on Android 14? There is currently a bug with GadgetBridge on Android 14 that causes firmware updates to not work. It'll be fixed in the next GadgetBridge release. There's a section in the 1.14 release notes about what you can do in the meantime.

AH!, okay. thanks

FintasticMan, 2024-01-07:
Duplicate of #1954.

closed 2024-01-07 by FintasticMan