InfiniTimeOrg/InfiniTime #1979: Gadgetbridge sees device in pairing mode, but doesn't respond to touch input

Ocean-OS, 2024-01-18:
### Verification

- [X] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

After updating my watch to 1.14, and repairing it because it wouldn't connect, Gadgetbridge can see the device, but will not respond to touch input and crashes. 

### What should happen instead?

After updating my watch, I would've hoped it would automatically reconnect to Gadgetbridge.

### Reproduction steps

1. Download the DFU file for 1.14, update the watch using Gadgetbridge
2. After the watch restarted, it wouldn't reconnect to my phone
3. Delete the watch in Gadgetbridge and try repairing it
4. It takes a little while, but the watch shows up
5. Tapping the watch in Gadgetbridge doesn't pair, any sort of input to Gadgetbridge does nothing, such as: Using the back swipe, pressing discovery settings, pressing stop scanning
6. After a minute or so, Gadgetbridge crashes

### More details?

I tried reopening Gadgetbridge, restarting my phone, updating my phone, force stopping Gadgetbridge, clearing Gadgetbridge's data. None of these things changed anything.

### Version

1.14

### Companion app

Gadgetbridge 0.76.1, Android 13, Samsung Galaxy A03s

FintasticMan, 2024-01-23:
This sounds like a GadgetBridge issue rather than an InfiniTime issue. Maybe you could open an issue on their issue tracker: https://codeberg.org/Freeyourgadget/Gadgetbridge/issues.

closed 2024-01-23 by FintasticMan

ndom91, 2024-02-23:
Having the same issue after updating to `1.14.0`. I haven't deleted the watch from Gadgetbridge yet though and tried readding. Didn't want to lose my history :sweat_smile:

I did notice though, that if you toggle on "Device Details" in gadgetbridge, that it sees the hardware and software version of the device (shows `1.14.0`) while trying to connect. Also the watch shows "Connected", but gadgetbridge seems stuck on "Connecting...".

Anyway, @Ocean-OS  did you ever open an issue on their repo? I couldn't find anything.

ndom91, 2024-03-10:
Coming back to say its working now for me now with 1.14 and latest Gadgetbridge 😅 

What i did was just completely unpair the infinitime in the android Bluetooth menu, and then just readd it from scratch in gadgetbridge. 

Idk maybe the Bluetooth mac had changed between fw updates 🤷‍♂️ anyway, works now. Shows "connected" in Gadgetbridge now too!