InfiniTimeOrg/InfiniTime #1984: Colors invert randomly

Rathmox, 2024-01-19:
### Verification

- [X] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

Colors inverted for no apparent reason

### What should happen instead?

Colors should stay the same

### Reproduction steps

?

### More details?

Looks similar to #1022 but did not happen when charging
Happened on 2 different Infinitime watches
[image]
[image]

### Version

1.14.0

KaffeinatedKat, 2024-01-21:
yea i'm really not sure why this happens, and i've seen it occur multiple times while working with the display driver. My best guess is there is some issue with how we are writing to the LCD control register, and accidentally (perhaps an overflow) writing the command to invert the colors. I'll have to do some investigation to be sure though

riban-bw, 2024-01-23:
I have seen this happen. Would a (temporary) work-around be to assert the flag to not invert colours when accessing a screen, e.g. watchface / info.

DEPSUT, 2024-01-31:
This has happened to me, but I know a way of reproducing this issue.

You need to turn on lower wrist to put the watch to sleep and go to the paint app, then just keep holding the screen and lower it until the screen turns off, and after you turn the screen on again, all the colors are inverted, I have attached a video showing it.

It may have something to do with the issue #1992 

https://github.com/InfiniTimeOrg/InfiniTime/assets/77130196/e7b14534-f661-42e3-80b6-b7260d171014

KaffeinatedKat, 2024-01-31:
I can reproduce this as well. This issue seems to also be the roots of the issue we're experiencing with #1869 (https://github.com/InfiniTimeOrg/InfiniTime/pull/1869#issuecomment-1915865847). There is some kind of issue with how the display is being controlled (SPI), which is causing unexpected behavior with the display

mark9064, 2024-01-31:
Having a consistent repro is amazing, thanks!!!!

Rathmox, 2024-04-08:
I managed to trigger this bug by turning off and on multiple times the watch. I never managed to reproduce it though.

JF002, 2024-05-01:
I've just merged @2046! This issue will hopefully not occur again :) 
Thanks to @mark9064 for their great work!

Saarsk, 2024-06-04:
> I've just merged @2046!

I think maybe you didn't intend to @ the user 2046 but instead refer to #2046? :)

JF002, 2024-06-09:
> I think maybe you didn't intend to @ the user 2046 but instead refer to #2046? :)

You're right, thanks for letting me know about this typo (and sorry to user 2046 for the unexpected ping :) ).

Ghoelian, 2024-12-02:
I just ran into this issue for the first time, running on the latest commit on main right now (commit 79ee886). Looks like it's not entirely fixed yet.

Ghoelian, 2024-12-02:
Actually sorry nevermind, I didn't validate the firmware so it reverted back to some other commit I don't remember.

Ghoelian, 2025-02-25:
Hey it's me again. The issue just resurfaced, this time on 1.15.0 (my own fork though).

Here's some pictures:

[image]

[image]

KaffeinatedKat, 2025-02-25:
what does your fork modiy? This bug should no longer be possible with the fix that was merged

- John Crawford

-------- Original Message --------

Ghoelian, 2025-02-26:
OTOH I added 2 settings related to vibration, and added a weather controller to a watchface. Could that have anything to do with this bug?

I'll look at the diff when I get home to be sure.

9cxndy, 2025-03-05:
I got hit by this as well. didn't happen when charging, lower wrist is turned on. I am running 728da0f4a08d327344aea with the background heartrate PR (#1718)

Ghoelian, 2025-03-13:
@KaffeinatedKat it doesn't look to me like any changes I've made could be the cause, but I don't really understand this issue tbh.

Are there any logs I could extract somehow, should this happen again?

If you want you can see all the changes I'm running here:

https://github.com/InfiniTimeOrg/InfiniTime/compare/main...Ghoelian:InfiniTime:ghoelian

PR #2046 (2024-05-01): SPI transaction hooks
This replaces the original technique for protecting the LCD data command pin (task notifications), which didn't work super well.

This is replaced with SPI transaction hooks, where when an SPI write request (transaction) is started, you can specify a hook that runs before the write and after it finishes. It's not a super elegant solution as the hook function has to be static (if someone with C++ knowledge wants to go nuts on that be my guest!) but it's pretty straightforward and simple at least. So for the LCD, the hook just sets the data command pin to the correct settin [... cut]