InfiniTimeOrg/InfiniTime #1994: Crash Screen (?) During Normal Use

Batcastle, 2024-01-29:
During regular use, I had the screen below come up. The watch was unresponsive, but I was able to reset it, and get it back to normal use.

[image]

Any idea what could cause this and if I should be worried?

mark9064, 2024-01-29:
This is the recovery firmware - it's accessible when the watch is starting by holding the button. If you hold the button for 7s it will reboot, and if you keep holding while it boots it'll enter the recovery firmware.

In the recovery mode you can flash a new firmware over bluetooth, or you can again hold the button to restart.

Perhaps the button accidentally got held for a while? Either way, there's nothing to worry about. Just make sure your firmware is validated in the settings page

Batcastle, 2024-02-03:
That's more than likely what it is. I just hadnt had to deal with the recovery firmware before so I wasnt sure what I was looking at.

Thanks for the help!

closed 2024-02-03 by Batcastle