InfiniTimeOrg/InfiniTime #2006: App menu stays at second screen

minacode, 2024-02-07:
### Verification

- [X] I searched for similar issues (including closed issues) and found none was relevant.

### Introduce the issue

1. Go to the app menu.
2. Go to the second screen. 
3. Press the button to return to the watch face.
4. Swipe to enter the app menu. 
5. You are still on the second screen.

### Preferred solution

I would expect the app menu to reset to the first screen when returning to the watch face.

### Version

1.14 (custom, but I doubt it matters)

tituscmd, 2024-02-07:
Could you specify your custom fork please? I've had this issue too ever since I've flashed my custom fork onto my watch, so it might actually be the custom fork that's causing this trouble.

minacode, 2024-02-07:
It's my simple calculator PR.

FintasticMan, 2024-02-07:
I have my suspicions that it's due to the compile-time apps, as that was the only time we've changed ApplicationList recently.

minacode, 2024-02-07:
Yes, probably. I don't understand yet where the state is saved. The information that its the second screen must survive outside the ApplicationList screen, because that gets destructed in between, right?
Must be somewhere in DisplayApp.

vkareh, 2024-02-07:
> I have my suspicions that it's due to the compile-time apps, as that was the only time we've changed ApplicationList recently.

I have this same behaviour directly in `main` without compile-time apps specified.

The issue is that the launcher is an "app" in itself with its own internal state. When you press the button, it pops the launcher from the stack and goes back to the watch face. When you swipe down, you're loading the launcher app again, but the screen loaded in the launcher is the same as it was.

The issue is likely the interaction between DisplayApp and ApplicationList. Maybe we need to ensure that the instance of ApplicationList gets destroyed when exiting it?

minacode, 2024-02-07:
> Maybe we need to ensure that the instance of ApplicationList gets destroyed when exiting it?

It doesn't?
The stack just holds the screen identifiers from the enum, but no whole screen instances, right? Where does the ApplicationList live on after returning to the watch face?

minacode, 2024-02-08:
The settings have an attribute `appMenu`. Could that be it? I don't have the time to look deeper into it right now.

vkareh, 2024-02-08:
Oh that's a good find, thanks @minacode!
I made a few changes and it seems to work with `appMenu`. I can try to submit a patch next time I'm at the computer.

vkareh, 2024-02-09:
Fix submitted: https://github.com/InfiniTimeOrg/InfiniTime/pull/2009

PR #2009 (2024-03-12): ApplicationList: Reset app menu screen when loading watch face
This prevents the application list from loading in the last used screen and instead goes back to the first screen whenever the watch face is loaded.

Fixes #2006
commit 3b4b5a51 (2024-03-12): ApplicationList: Reset app menu screen when loading watch face

This prevents the application list from loading in the last used screen
and instead goes back to the first screen whenever the watch face is
loaded.

Fixes #2006
closed 2024-03-12 by JF002