InfiniTimeOrg/InfiniTime #2008: State of GFX?

mark9064, 2024-02-08:
GFX within InifniTime (https://github.com/InfiniTimeOrg/InfiniTime/tree/main/src/components/gfx) isn't used by anything anywhere as far as I can tell?

Could anyone explain the history of this? It's taken me a while to realise it's not part of the OS render pipeline at all and its existence is pretty confusing!

Maybe it is time for it to be removed if it is legacy? At least for me dead code adds to the effort to understand InfiniTime and slows development down. Git will always remember it if we want to refer back

JF002, 2024-02-09:
I implemented the GFX class at the very beginning of the project, before we replaced it by LVGL. It's very basic, but allowed me to design and optimize the SPI and LCD drivers.

I thought that it was still being used in the recovery firmware but if it's not used, I'm 100% OK to remove it : I don't like to maintain dead code either!

KaffeinatedKat, 2024-02-09:
if it is being used by the recovery firmware, it should be changed to just use LVGL like everything else and be removed. Don't see much of a reason to keep and maintain some legacy code just for the recovery firmware if it can just use LVGL like the rest of InfiniTime

JF002, 2024-02-11:
> if it is being used by the recovery firmware, it should be changed to just use LVGL like everything else and be removed. Don't see much of a reason to keep and maintain some legacy code just for the recovery firmware if it can just use LVGL like the rest of InfiniTime

I do not agree with this : We must keep the code of the recovery firmware as simple as possible to reduce the risk of critical bugs (less code and less complexity -> less bugs). The recovery firmware does not integrate a complex UI, it simply displays a logo full screen, and it does not need the whole LVG

mark9064, 2024-02-11:
Thanks for the discussion and review :)

I'll close this now

closed 2024-02-11 by mark9064