InfiniTimeOrg/InfiniTime #2012: Watch freezes after some notifications

everypizza1, 2024-02-11:
### Verification

- [X] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

I received a notification and the touch screen froze for about 1 minute

### Reproduction steps

It doesn't consistently happen, but some things I've noticed:
* Only happens on notifications with emoji or with text in non-English languages.
* Only affects touchscreen, notification still scrolls.

### More details?

Only seems to affect notifications from YouTube for live streams and foreign language notifications from apps like holoplus.foregin

### Version

v1.14

### Companion app

Gadgetbridge 0.78.0

minacode, 2024-02-11:
InfiniTime does not support all characters. Gadgetbridge might handle this somehow, but I don't know how for sure. 

1. Do you have an example of a notification that causes the bug? 
2. Is there a way to find out what text Gadgetbridge is actually sending to the watch?

everypizza1, 2024-02-11:
One almost certain way that it's triggered is when someone I have subscribed to does a YouTube livestream. Gadgetbridge seems to be sending the notifications from YouTube with :red_circle: to replace the 🔴 in the notification. I don't have any notifications currently on the watch, but this is one from my notification history that caused it:
[Image]

(I did this through email so it might not work)

sbe-arg, 2024-02-13:
I'm noticing this as well. Specially with emoji notifications from whatsapp and signal.

Also using gadgetbridge.

minacode, 2024-02-15:
Have you tried other companion apps?

FintasticMan, 2024-02-15:
Could you try finding a reproducable case by trying it out in GadgetBridge's Debug menu? That allows you to send test notifications.

sbe-arg, 2024-02-15:
> Could you try finding a reproducable case by trying it out in GadgetBridge's Debug menu? That allows you to send test notifications.

Been trying to reproduce it without any luck a message that might lag when coming from the app not necessarily lags when triggered via debug.

But ill keep checking

everypizza1, 2024-02-15:
I also can't reproduce it in Gadgetbridge.

everypizza1, 2024-02-15:
I also can't reproduce this in Gadgetbridge.

everypizza1, 2024-02-16:
Just happened again with this notification:
[image]

minacode, 2024-02-17:
This was with Gadgedbridge? 
If yes, is there any chance of finding out what Gadgetbridge actually sends to the watch? 
It does transform the notification text to something InfiniTime should be able to show, right?

everypizza1, 2024-02-17:
Gadgetbridge will always convert emojis to a description of them, like red_dot for 🔴, surrounded by colons. For example: `:red_dot:`.

minacode, 2024-02-17:
Ok, but can you somehow reproduce what it sent in the case of your faulty notification?

everypizza1, 2024-02-18:
I've tried to reproduce many times with no success. It has just happened again though.
Watch:
[image]
Phone:
[image]

sbe-arg, 2024-02-18:
Do you have the heart rate monitor on? I noticed it only happens in my case when is on if of I don't have the freeze.

minacode, 2024-02-18:
@everypizza1 
The notification contains a longer dash character, like "—".
In the title of your notification on the watch, the seems to be a strange character displayed right after "Pancakes" where the dash should be. I wonder what happened there.

@sbe-arg
Oh, no 😄
Let's see where this goes. Maybe there is another race somewhere. The heart rate has its own task if I remember correctly.

everypizza1, 2024-02-22:
> @everypizza1 
> The notification contains a longer dash character, like "—".
> In the title of your notification on the watch, the seems to be a strange character displayed right after "Pancakes" where the dash should be. I wonder what happened there. …

I just realized that it skipped the long dash and went straight to the artist name. Apparently all Genius notifications have a long dash, I just tested with other songs. However others didn't cause it for some reason.

Self-Perfection, 2024-03-04:
I probably have the same issue but with some variations:

In these hang state these function does not work:
* screen swipes (no reaction)
* single button press to go one screen back

But these function keep working:
* Seconds hand on analog watch face keeps ticking
* Longer button presses to get to clock or firmware version or reboot
* Double button press to switch to notifications screen

In my case waiting does not unfreeze watch. I have to reboot.

It have not been happening with previous firmware version, but have happened about 5-7 times with 1.14.0.

I have no idea what triggers freezing.

FintasticMan, 2024-04-30:
Hmm, good observation. Maybe there's a race somewhere triggered by trying to go to sleep.

timteichmann, 2024-05-21:
Not sure if any further confirmation is required. But I manage two daily driver PineTimes and have experienced regular freezes (every other day) that would require a reset with the setting "raise to wake" and "lower to sleep" enabled.

I can confirm that disabling "lower to sleep" as per @everypizza1 's comment (Cheers for posting that here) does resolve the issue. This seems to be the culprit.

copyrights, 2024-06-10:
> I probably have the same issue but with some variations:
>
> In these hang state these function does not work:
>
>     * screen swipes (no reaction)
>
>     * single button press to go one screen back
>
>
> But [...]

I have the same behavior. From what I have observed so far it happens like this
* A notification pops up
* the notification disappears regular after timeout, but display is still on.
* **Any touch interaction (or maybe other interaction too) triggers bug**
* If unnoticed battery drains fast
* Only reset (very long button press) helps

everypizza1, 2024-06-10:
@copyrights Do you have lower-to-sleep turned on?

copyrights, 2024-06-11:
> [Verfolgungsbild][https://github.com/notifications/beacon/AABVPAO2QD5ZU5AAK4DXUMTZGYG2TA5CNFSM6AAAAABDDU2TT2WGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTUAWO7JG.gif]

Yes, Settings -> Wake Up -> Lower Wrist was on.  I disabled it now and will see if it mitigates this issue.

sbe-arg, 2024-06-11:
It hasn't happened to me in a while now no changes other then regular updates of gadgetbridge.