InfiniTimeOrg/InfiniTime #2021: Watch freezes and reboot after interacting with external resources

liamcharger, 2024-02-13:
When interacting with external resources (through InfiniLink), and then changing a setting, the watch will freeze, and then reboot. 

The issue appears to be when InfiniLink uses an LS function to make a list with all the files in the selected directory.  

The code used in InfiniLink to fetch the external resources has been triple-checked and does not appear to contain any potential causes for this issue, so it's possible the issue resides in InfiniTime.

JF002, 2024-02-13:
Do you know of anyway to reproduce this, preferably without an iOS device? Is this reproducible with another companion app like Amazfish or ITD, for example?

liamcharger, 2024-02-13:
I haven't used Amazfish or ITD yet, I'll try installing one of them and let you know.

everypizza1, 2024-02-26:
It's not just InfiniLink, it also happens with https://infinitimeexplorer.netlify.app/

liamcharger, 2025-03-24:
This issue is still occurring in 1.15.

Also, I found another issue that has been around for a while: #1579