InfiniTimeOrg/InfiniTime #2024: Occasional Incorrect Dates Displayed in the Weather forecast

JenJell, 2024-02-21:
### Verification

- [X] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

The weather forecast sometimes has the dates outset by one.

### What should happen instead?

The dates should not be offset in any direction.

### Reproduction steps

If you can't currently see the issue, you will need to adjust the UNIX timestamp being sent to some other time of the day.

### More details?

For the past few days, I have been testing the new weather app in InfiniTime. However, I have noticed that the forecast does not always show the correct dates. I have checked to make sure I am sending the local UNIX timestamp using InfiniLink, but, everything there seems to be correct, so I suspect this is probably a bug in InfiniTime.

### Version

main

### Companion app

Development Version of InfiniLink

vkareh, 2024-02-21:
Hi @JenJell - can you give an example of what you mean by dates being offset?

JenJell, 2024-02-21:
Hello @vkareh - sorry for the confusion. Hopefully this will help clear up what I am trying to convey. Let's say, if it is Tuesday at 3:00 AM in Tokyo, the weather app will display Tuesday as the first day in the 5-day forecast. The problem here is, at this point in time, it should display Wednesday instead. Hope this helps.
[image]