InfiniTimeOrg/InfiniTime #2030: Turn off accelerometer in night mode

JustToby22, 2024-02-27:
### Verification

- [X] I searched for similar issues (including closed issues) and found none was relevant.

### Introduce the issue

In night mode, I realised that while the PineTime doesn't react to any accelerometer-based "display on" events, it does react to the "lower wrist" event.
I have to say, I haven't read the code, but it seems to me that the accelerometer is still active and reporting its state during night mode. Since many people, as I do, likely use Night mode to preserve power, it seems illogical to keep the accelerometer running while it's on.
One could argue that it would still be necessary for step counting, however. Personally, I'd take power savings over accurate step count while I'm trying to save power/go to sleep.

### Preferred solution

Disable the accelerometer in night mode.

### Version

1.14.0

FintasticMan, 2024-02-27:
I think that the accelerometer does indeed not turn off in night mode, but you wouldn't be able to tell by lower to sleep still working. This is because the accelerometer should only turn off when the screen is off.

L3br4nd, 2024-06-19:
What about people who want to use the watch for sleeptracking? 
They would still need the accelerometer. 
In that case there should be an option to keep it on.

FintasticMan, 2024-06-19:
The accelerometer stays awake if there is a device subscribed to the motion service. This means that if you're using an app on your phone to track sleep, it won't turn off. We'll make sure to not break that while fixing this bug.