InfiniTimeOrg/InfiniTime #2033: Can't connect to Gadgetbridge after Firmware 1.14 update.

FreeJulianAssange, 2024-03-01:
### Verification

- [X] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

Bluetooth fails to connect after 1.14 firmware update.

### What should happen instead?

Watch connects to Gadgetbridge as usual.

### Reproduction steps

1. Update Pinetime using Gadgetbridge File installer (Firmware 1.14)
2. Watch displays Bluetooth connection Symbol but Gadgetbridge just shows "connecting" without ever connecting.
3. Resetting the firmware back to 1.13, Gadgetbridge immediately connects to the watch. 

### More details?

I have two Pinetimes, both update without issue to 1.14 but then fail to connect to Gadgetbridge after the update.
I've tried rebooting both devices which usually solves this type of issue without success.

Unfortunately I'm unable to provide any logs as my device is not rooted (Android 13).

### Version

V1.14

### Companion app

Gadgetbridge 0.78.0 (baaee8f58)

SteveAmor, 2024-03-01:
I always end up having to pair again, after upgrades. So I remove the device from GB and Android and start afresh.

z3t0, 2024-03-01:
I was able to get it to work by disabling and enabling bluetooth on the watch, and then disconnect and reconnect on the gadgetbridge. Hope it helps

FreeJulianAssange, 2024-03-02:
> I always end up having to pair again, after upgrades. So I remove the device from GB and Android and start afresh.

I had some time just now to play around with the watch again, removing the device & re-pairing solved the connection issue but I've stumbled upon a new issue.

When I Validate the Firmware, the watch resets itself and returns back to Firmware v1.13.

b02860de585071a2, 2024-03-10:
> I always end up having to pair again, after upgrades. So I remove the device from GB and Android and start afresh.

What has worked for me on 1.14 is:

0. (if previously connected) Delete/forget watch in GB/Android settings
1. Connect to GB first _without any bonding_
2. Toggle watch BT off
3. Open pairing menu in Android settings
4. Toggle watch BT back on
5. Pair as normal through Android flow
6. Tap on watch in GB to reconnect

Pairing in Android first, attempting to bond in GB when adding the watch, or any other combination of things all seem to fail. 

I'd guess there's some conflict between how GB and the OS handle device pairing, but that's beyond my expertise.

JF002, 2024-03-12:
I also noticed that I sometimes have to remove my watch from the Android settings before being able to connect with Gadgetbridge after an update. I haven't had the opportunity to figure that out yet, unfortunately.

I'll close this issue since it appears to be resolved/worked around.

@FreeJulianAssange Feel free to open another post related to the validation issue you experienced.

closed 2024-03-12 by JF002