InfiniTimeOrg/InfiniTime #2069: Not reconnecting to Gadgetbridge

KedynsCrow, 2024-06-01:
### Verification

- [X] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

When the watch disconnects from Gadgetbridge, then reconnects it no longer reports batter level or sends notifications to the watch.

### What should happen instead?

After disconnecting I expect the watch to automatically reconnect and keep working as it did before.

### Reproduction steps

I connect the watch to the Gadgetbridge app on the phone.  It shows the watch's battery level, the watch receives notifications from the phone and everything works as I expect.  Then I put the phone down and walk away.  It disconnects as I would expect.  When the phone comes back into proximity of the watch the Gadgetbridge app says it has connected again, but no longer shows the battery level of the phone, the step count is out of sync, and the watch no longer receives notifications.  This state continues until I go to the Gadgetbridge app, disconnect the watch, and then reconnect it.

### More details?

I have a Samsung phone running Android 13.  The watch had this issue with Infinitime 1.11 and it continued when I updated to 1.14.  The Gadgetbridge app has all permissions allowed from the phone.  Gadgetbridge is set to "Connect to Gadgetbridge device(s) when Bluetooth is turned on" and in the device specific settings it is set to "Auto reconnect to device."

### Version

1.14

### Companion app

Gadgetbridge 0.79.1a

SteveAmor, 2024-06-02:
For me, this happens using simple pairing but works as expected using secure bonding with a pin.

pbone64, 2024-06-02:
Same for me. My connection was stable once I paired

qw3r3wq, 2024-06-25:
Hi all,

looks like a bit related:
https://codeberg.org/Freeyourgadget/Gadgetbridge/issues/3741 . This issue describes the same with Android 14 Pixel (Stock) phone.
Also, I do have issues with Nothing Phone 2 (Android 14, Stock).

Gadget bridge 0.80.0

DavisNT, 2024-07-04:
@KedynsCrow, @qw3r3wq, @pbone64 Can you please try this procedure (please do it precisely) and let me know whether it resolves the issue?

Before beginning this procedure, please ensure a properly working version of InfiniTime has been installed on the watch and [validated](https://github.com/InfiniTimeOrg/InfiniTime/blob/main/doc/gettingStarted/updating-software.md#firmware-validation).

1. In Gadgetbridge tap on the 3 dots on InfiniTime tile, choose _Delete Device_ and confirm the deletion by tapping _Delete_.
2. In Gadgetbridge tap on the plus (add device) button, select InfiniTime and choose _Pair_.
3. A 6 digit code should appear on the watch and a prompt should appear on the phone/tablet. Type the 6 digit code from the watch in the prompt on phone/tablet and tap _OK_.
4. The InfiniTime device should now be connected in Gadgetbridge (in Gadgetbridge the InfiniTime tile should display status "Connected").
5. On watch press the side button to dismiss the screen with the 6 digit code.
6. On watch open settings, open _Bluetooth_ and select _Disabled_.
7. On watch press the side button 3 times (with around 1 second between the presses) to go back to the watch face screen (this part is extremely important!).
8. On watch open settings, open _Bluetooth_ and select _Enabled_.
9. On watch press the side button 3 times (with around 1 second between the presses) to go back to the watch face screen (this part is extremely important!).
10. On watch press and hold the side button (for around 8 seconds) until the PineTime logo (a white pine cone) appears - release the side button immediately when the PineTime logo appears.

pbone64, 2024-07-05:
@DavisNT I have performed your method and it seems stable so far, though when I had connection issues they would take a couple of days to start happening each time that I connected to my watch.

To be clear, though, my connection has been stable ever since pairing to my pinetime (instead of connecting without pairing), regardless of what steps I perform after pairing. Are your steps meant to provide a stable connection? In my experience it has been enough to just pair.

timaw7, 2024-07-19:
Experiencing this as well.  Just to note with this, there's a loss of data like steps.  I only notice every few days myself.  A hard reset of the watch reconnects, but by then all data is lost.

I wonder if there should be some connection status on the watch itself that can inform the user outside of pulling open the app.

pbone64, 2024-07-19:
> Experiencing this as well. Just to note with this, there's a loss of data like steps. I only notice every few days myself. A hard reset of the watch reconnects, but by then all data is lost.
>
> I [...]

There is, at least on the terminal watchface. I believe the analog and digital faces have a Bluetooth symbol when connected, but I'm not certain as I  haven't used them a while.

Regardless, at least in my experience, when this bug occurs the watch believes itself to be connected while the companion app (gadgetbridge in my case) believes it to be disconnected.