InfiniTimeOrg/InfiniTime #2073: What watch faces currently support weather?

sbe-arg, 2024-06-06:
### Verification

- [X] I searched for similar issues (including closed issues) and found none was relevant.

### Introduce the issue

None of the faces in 1.14 seem to support weather?

I'm using gadget bridge with quickweather and can see the weather info in gadgetbridge debug settings but none of the faces show the weather info.

### Preferred solution

An up to date state of the watch faces and their features/compatibility by version.
maybe a table in the reademe for feature by version support map.

### Version

1.14

sbe-arg, 2024-06-07:
Adding this for reference https://wiki.pine64.org/wiki/Infinitime-Weather 

it works on face PineTimeStyle

tituscmd, 2024-06-07:
It also works on the Digital watchface

sbe-arg, 2024-06-07:
> It also works on the Digital watchface

How do I enable weather in the digital face? Can't figure it out.

tituscmd, 2024-06-08:
> > It also works on the Digital watchface
> 
> How do I enable weather in the digital face? Can't figure it out.

When using a companion app that allows weather data to be sent to the watch, it will appear on the Digital face automatically.

sbe-arg, 2024-06-08:
> > > It also works on the Digital watchface
> > 
> > How do I enable weather in the digital face? Can't figure it out.
> 
> When using a companion app that allows weather data to be sent to the watch, it will appear on the Digital face automatically.

Does not seem to be the case in 1.14 it works only in the pinetimestyle for me

FintasticMan, 2024-06-08:
That feature was added since 1.14.

tituscmd, 2024-06-08:
I think you have to use the very latest version of 1.14 from GitHub directly to get weather.

alexllama, 2024-06-09:
1.14 from github worked for me

JF002, 2024-06-09:
The weather in the digital watch face was added with [this PR](https://github.com/InfiniTimeOrg/InfiniTime/pull/1856). It'll be released in InfiniTime 1.15 (it's not available yet in the [official 1.14 release](https://github.com/InfiniTimeOrg/InfiniTime/releases/tag/1.14.0).

MorsMortium, 2024-09-13:
I work on a watchface, that also supports it. Feel free to check it out: https://codeberg.org/MorsMortium/GTKWatchFace
[image]

closed 2026-01-08 by sbe-arg