InfiniTimeOrg/InfiniTime #2074: Whats the best way to track what and when the next release

sbe-arg, 2024-06-09:
Any way to follow up next release overview, eta in projects, issues, etc?

Matrix?

JF002, 2024-06-09:
We do not have any specific roadmap and/or planning : we are working on the project when we have the time to do so : working on new feature, improve the current code, do some maintenance, review and merge PRs,...

You can have an overview of the PR that we have already merged for the next release in the [milestone](https://github.com/InfiniTimeOrg/InfiniTime/milestone/15) and those that we plan on reviewing/merging for the next release. Note that this is a moving target : we often add/remove PR from the milestone depending on the reviews and feedback from other developers.

You can also engage with developers and users in the community [PineTime chat rooms](https://pine64.org/community/) (look for **PineTime** and **PineTime Development**).

sbe-arg, 2024-06-09:
Thanks for the insights.