InfiniTimeOrg/InfiniTime #2075: Add active alarm indicator to watch face

sammyj85, 2024-06-10:
### Verification

- [X] I searched for similar issues (including closed issues) and found none was relevant.

### Introduce the issue

Twice or three times I have accidentally turned off my alarm after setting it.

The only way to check if the alarm is still set is to enter the alarm screen, and observe that the switch is on, and potentially risk accidentally touching the switch (clumsy) as you press the button to exit the screen. I end up checking this three times after I set it, to make sure it's still enabled.

### Preferred solution

A way to be very confident that the alarm is set, is to put an alarm icon indicator on the watch face. Most watches I've previously had did have such an indicator.

This obviously involves trying to see where there is space to put it on each face.

### Version

v1.14

kyeastmood, 2024-06-18:
I agree that an alarm indicator on the main screen would be useful.

But to solve the underlying problem there should be two separate screens:
1. For configuring the alarm (enable/disable/set time/set recurrence)
2. For switching off or snoozing a ringing alarm

Currently the same screen is used for both so when the alarm is ringing it's much too easy to disable it completely instead of just switching it off.

sammyj85, 2024-06-19:
I agree splitting it into two screens would be an improvement. But that would probably be better tracked in a separate issue.

kyeastmood, 2024-06-21:
You're right @sammyj85, I will create a separate issue for the alarm screens.

margamanterola, 2024-07-07:
I've been thinking about this. I think the alarm icon could be added to the StatusIcons that are shown in the different watch faces, to indicate that the alarm is set.

Not all watch faces use them, but I've been doing a bit of refactoring to the StatusIcons for a watch face I'm working on, and I believe it's possible to make more watch faces use them by making some aspects changeable via function calls.

MorsMortium, 2024-09-13:
Hi! After reading this issue, I have added an active alarm indicator to the watchface I am working on. Feel free to check it out. https://codeberg.org/MorsMortium/GTKWatchFace
[image]