InfiniTimeOrg/InfiniTime #2081: Fix Stopwatch Side Button Behavior

RageGamerBoi, 2024-06-17:
### Verification

- [X] I searched for similar issues (including closed issues) and found none was relevant.

### Introduce the issue

This issue relates to two problems related to side button behavior that can and should all be fixed together:
1. The stopwatch is stopped by the side button but cannot be restarted with the side button.
2. There is no way to exit the stopwatch and let it run in the background, as the back button stops the stopwatch.

### Preferred solution

1. Allow the stopwatch to be started by the side button. It might be better to use a double click gesture, which can also be used for the lap function once the stopwatch is started.
2. Allow the ‘press and hold’ gesture to exit the stopwatch without stopping it, so you can check the time, etc.

### Version

v1.14.0

mark9064, 2024-06-18:
Another issue that the stopwatch stops when you release the button rather than press it. But this would require input events to be refactored to allow apps to process button up/down themselves

dozadoesit, 2024-12-27:
so if the single use button starts and stops the the stop watch how do you exit the stop watch? I know you can swipe down to get out of it but this might not be intuitive for all users. I think if we go this route we might also want to add some sort of back button to the app to exit.

mark9064, 2024-12-27:
2) is solved by #2141 at least