InfiniTimeOrg/InfiniTime #2082: Use separate screens for switching off ringing alarm and for alarm configuration

kyeastmood, 2024-06-21:
### Verification

- [X] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

Its too easy to disable the alarm completely instead of just switching it off when the alarm is ringing

### What should happen instead?

There should be separate screens for alarm configuration and for swithining off a ringing alarm

### Reproduction steps

Currently, when the alarm is ringing you can push the red button to switch it off (so the button will turn green). But then it's easy to push the button again by accident (especially when being in bed and not looking at the watch) which will disable the alarm completely and it won't ring the next day.

### More details?

Solution:  use two separate screens for dealing with the alarm:
1. For configuring the alarm (enable/disable/set time/set recurrence)
2. For switching off or snoozing a ringing alarm

### Version

1.14

kyeastmood, 2025-01-04:
Hi @ljahn, thank you for working on this problem! It is the most annoying issue in InfiniTime for me.

tituscmd, 2025-02-06:
If I understand correctly, this should be fixed by #2211

ljahn, 2025-02-06:
Yes, it is.  
This issue wasn't auto closed when my PR was merged, because the fix for this was the result of dealing with another issue. Now there are two separate screens: One for configuration, and one when alerting, where one can only press Stop, nothing more.  

The only remaining this that is not covered is the mention of a snooze function. That is not implemented yet, but could well be its own issue/feature request.

mark9064, 2025-02-07:
Agree, closing this as solved as the headline issue is resolved. For snooze discussion I think a new issue / discussion post would be best

closed 2025-02-07 by mark9064