InfiniTimeOrg/InfiniTime #2084: Firmware update with pinetime-mcuboot-app-dfu-1.14.0.zip and GadgetBridge fails

luto65, 2024-06-26:
### Verification

- [X] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

GadgetBridge reports "connection problems"

### What should happen instead?

GadgetBridge should connect

### Reproduction steps

Installed GadgetBridge 0.80.0 on a HUAWEI
PineTime hardware
downloaded on the HUAWEI the pinetime-mcuboot-app-dfu-1.14.0.zip

### More details?

I received the PineTime and with my iphone and InfiniLink i connected to it and updated with the latest version 1.14.0
Then I wanted to try out how the updating of the pinetime through DFU would work and therefore i installed GadgetBridge on the HUAWEI 
I tried to connect several times , and while the PineTime shows up in the BLE devices, often the connection last few seconds and then disconnect.
A only time I was able to connect, then started the upload of the firmware but then the connection broke again.

### Version

v.1.14.0

### Companion app

GadgetBridge v 0.80

DavisNT, 2024-07-04:
@luto65 Can you please try this procedure (please do it precisely) and let me know whether it resolves the issue?

Before beginning this procedure, please ensure a properly working version of InfiniTime has been installed on the watch and [validated](https://github.com/InfiniTimeOrg/InfiniTime/blob/main/doc/gettingStarted/updating-software.md#firmware-validation).

1. In Gadgetbridge tap on the 3 dots on InfiniTime tile, choose _Delete Device_ and confirm the deletion by tapping _Delete_.
2. In Gadgetbridge tap on the plus (add device) button, select InfiniTime and choose _Pair_.
3. A 6 digit code should appear on the watch and a prompt should appear on the phone/tablet. Type the 6 digit code from the watch in the prompt on phone/tablet and tap _OK_.
4. The InfiniTime device should now be connected in Gadgetbridge (in Gadgetbridge the InfiniTime tile should display status "Connected").
5. On watch press the side button to dismiss the screen with the 6 digit code.
6. On watch open settings, open _Bluetooth_ and select _Disabled_.
7. On watch press the side button 3 times (with around 1 second between the presses) to go back to the watch face screen (this part is extremely important!).
8. On watch open settings, open _Bluetooth_ and select _Enabled_.
9. On watch press the side button 3 times (with around 1 second between the presses) to go back to the watch face screen (this part is extremely important!).
10. On watch press and hold the side button until the PineTime logo (a white pine cone) appears (for around 8 seconds) - release the side button immediately when the PineTime logo appears.

franzcor, 2024-08-03:
I've seen this problem reported multiple times and there is a similar issue on the Gadgetbridge repo too. I am also affected by the same issue unfortunately, without being able to solve it.

JF002, 2024-08-18:
This kind of issue is not easy to diagnose as there are a lot of moving parts. The issue could be caused by the PineTime, InfiniTime, the companion app, the BLE stack on the host device, the hardware of the host device,...

This is not a formal proof, but the fact that InfiniLink works makes me think that the issue does not necessarily come from InfiniTime. Gadgetbridge also works fine on my Samsung S21 smartphone.

jds11111, 2024-09-05:
Here is more hardware variation, and strange but reproducible results, in case it gives you any new clues. 

I just updated to 1.14.0 from 1.7.0, which created similar problems. My phone is a Google Pixel 6a running latest GrapheneOS. Gadgetbridge is 0.81. After inability to connect, I rebooted the watch, rebooted the phone, deleted device from GB, and unpaired from phone. I also did the disable, 3 presses, reenable thing. Pairing New Device sees the watch. However, it fails to bond. It first just darkens the screen, which eventually returns "Bonding with %1$s failed immediately".

I then tried with a OnePlus 8 running LineageOS 21 and GB 0.81.0, but got the same results.

I tried a Galaxy Pixel Tablet running GrapheneOS, same as above. It failed

Then tried with a Samsung Galaxy Tab running Android 9, which paired. I deleted the device from GB. 

Now, when I tried again, I could pair with the Pixel Tablet, but not the Pixel 6a. Repeated attempts showed the same results: after pairing once with the Samsung Tablet and then deleting it, I could pair it with the Pixel Tablet.

Finally, I discovered that I could pair it with the Pixel 6a, but only if it is not made a companion device.

franzcor, 2024-12-14:
> This kind of issue is not easy to diagnose as there are a lot of moving parts. The issue could be caused by the PineTime, InfiniTime, the companion app, the BLE stack on the host device, the [...]

I installed 1.14.0 on Jan 4 2024. Since then I was not able to connect anymore, despite upgrading Gadgetbridge.
This issue was not resolved and I cannot upgrade to 1.15
What I know is that I didn't upgrade my Android (it's stuck to latest available android 11) and I tried to revert to old Gadgetbridge which worked at that time. No luck.
So all the clues point to 1.14.0. I honestly don't know what to do as last time I tried to pair with a linux computer I couldn't even start the program (I don't remember which one it was)