InfiniTimeOrg/InfiniTime #2087: Wake-up Alarm does not sound in Sleep mode

SukiiPooru, 2024-07-02:
### Verification

- [X] I searched for similar issues (including closed issues) and found none was relevant.

### Introduce the issue

The  set wake-up alarm does not work in the [blue] sleep mode, it only works in the [green] alerts mode.
If I am sleeping I do not wan't random notifications, hence I put in [ blue ] sleep mode, but I still want the wake-up alarm to sound/vibrate/activate. 
In the [red ] stop mode it is obvious I don't want any notifications or alarms to sound, but in [Blue] wakeup, I want to get up in time (for appointments etc)

### Preferred solution

Blue  sleep mode needs a check for allowing alarm.

### Version

1.14.0

mark9064, 2024-07-02:
The alarm works fine for me in sleep mode. Could you detail the exact steps you do which causes the alarm not to fire?
Here's what I did to test
- Set alarm for the next minute
- Switch notification mode to sleep
- Turn the screen off by tapping the button
- Observe that the alarm fires and wakes the device

SteveAmor, 2024-07-02:
I set the alarm for 2 mins in the future. Didn't check with the i. Set to blue sleep. Turned off with button. Alarm fired after 2 mins. V1.13.0 5f3acfad

SukiiPooru, 2024-07-04:
> steps you do which cause the alarm not to fire?

foolishly I didn't think of the 2min ahead test. It worked ok. My
conclusion for the none function alarm was a quirk of crossing time
zone border. Sometimes smart phone doesn't update time for up to many
hours later [ poor signal / not updating changed tower band frequency],
phone must have updated pinetime across the alarm time, hence it didn't
sound. Appologies.

mark9064, 2024-07-04:
No worries at all, feel free to close this issue if everything is resolved now

mark9064, 2024-11-17:
Closing this after inactivity. If you encounter this again please feel welcome to open a new issue. Thanks for the report :)

closed 2024-11-17 by mark9064