InfiniTimeOrg/InfiniTime #2093: BLE bond deleted after 2 reboots without connectivity to central

DavisNT, 2024-07-08:
### Verification

- [X] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

After 2 reboots bond was deleted

### What should happen instead?

Bond is preserved during any number of reboots

### Reproduction steps

1. Turn off bluetooth on the central (phone etc.)
2. Reboot PineTime
3. Reboot PineTime
4. Bond now is gone

### More details?

This behavior is counter-intuitive and likely most InfiniTime users are not aware of this behavior.
Likely this might also have negative impact on security - at least some companion apps (e.g. Gadgetbridge 0.79) will continue to function as before after the bond is deleted on the watch (whether any encrypted pairing is done in this case or all data is transferred unencrypted is an open question).

This is by design (see #796 and 51f62ad096024d45c0274267a90d6adfe58c7c6d ), but why the bond is removed after 2 reboots?
Does existence of the bond have any negative effects? Maybe this could/should be converted into a button under Bluetooth settings (or something like that)?
@evergreen22 Thanks for the BLE bonding functionality! Maybe you know answers to the questions above?

### Version

1.14.0

### Companion app

Gadgetbrigle, but this will also apply to others

evergreen22, 2024-07-09:
"[It's not a bug it is a feature](https://www.wired.com/story/its-not-a-bug-its-a-feature/)"

To clarify your issue report, the bond is only deleted if there are two consecutive reboots ***without*** a re-connect to the companion. Each time Infinitime reconnects with the bonded device the "counter" is reset to zero.

Once the watch and companion are bonded no other device may connect to Infinitime (see [bluetooth spec](https://www.bluetooth.com/specifications/specs/core-specification-5-0/) - TL;DR Infinitime will not accept a new connection attempt from an unbonded device).

Consider the scenario where Infinitime has bonded but your companion has lost its bond (maybe you deleted them, maybe your bluetooth stack wigged out, maybe venus is rising in scorpio). The only way to re-bond your watch in this case would be to erase the bond using a hardware programmer (at least this was true at the time of #796 - my personal version has diverged quite a bit since then to include full privacy and encryption as well as dozens of personal customizations - I haven't followed Infinitime closely lately)

IMO the bond delete feature/design should definitely be (better) documented behavior.

evergreen22, 2024-07-24:
Sorry I wasn't clear in my previous reply. I was trying to say that Infinitime only supports one connection at a time. Of course, BLE allows multiple devices to be bonded simultaneously.