InfiniTimeOrg/InfiniTime #2096: PineStore announcement : the current flash chip is EoL, a new one will be used for future PineTime batches

JF002, 2024-07-21:
### Verification

- [X] I searched for similar issues (including closed issues) and found none was relevant.

### Introduce the issue

I've just received this information from TL Lim from PineStore : the current SPI flash memory chip (XTX XT25F32B) is EoL (end of life) and will be replaced by another one : BY25Q32ESWIG. 

Here is the datasheet : [BY25Q32ESWIG-03.A.02.00372.pdf](https://github.com/user-attachments/files/16323586/BY25Q32ESWIG-03.A.02.00372.pdf)

The only change expected on the software point of view are the JEDEC IDs :

| Chip | Manufacturer | Memory type | Density |  
|---------|---------------------|---------------------|-------------|
| XTX XT25F32B    | 0x0B | 0x40 | 0x16 |
| BY25Q32ESWIG | **0x68** | 0x40 | 0x16 |

PineStore asked me to provide them with a new firmeware that supports this new chip.

**Feel free to forward this announcement to all firmware developers that might be interested to support this new flash chip!**

### Preferred solution

At first sight, no change should be necessary in InfiniTime, since it does not check the IDs (it only logs the value, but does not fail if they are not the ones that were expected).

However, the SPI flash driver in the [bootloader](https://github.com/InfiniTimeOrg/pinetime-mcuboot-bootloader) will most probably fail since those IDs are specified and checked at boot time. 

Since the bootloader is a very critical part of the project, please feel free to cross-check my analysis and also test and provide feedback when a new version of the bootloader will be available :)

### Version

1.14.0

closed 2024-08-05 by JF002

SteveAmor, 2024-08-28:
Do you think there are going to be any issues with people who receive one of these new PTs and flash the bootloader to use WASPOS and then use the process/file to flash back to Infinitime?

JF002, 2024-09-01:
@SteveAmor This is a very good question! According to TL Lim and my observations, the new chip behaves exactly the same way than the old one, so no code change should be needed except if the bootloader/reloader checks the chip IDs of the memory chip. 
Do you know if this announcement has reached the WaspOS development team?

PR #2097 (2024-08-05): Display the SPI flash JEDEC IDs in SystemInformation.
This is needed since a new memory chip will be used in future batches of PineTimes.

Fixes #2096