InfiniTimeOrg/InfiniTime #2101: Sleep mode doesn‘t disable wake up actions anymore.

tituscmd, 2024-08-23:
### Verification

- [X] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

Using the newest build with the aod, the sleep mode doesn‘t seem to disable raise to wake or tap to wake anymore.

### What should happen instead?

The watch should only wake up via a button press when in sleep mode

### Reproduction steps

Use latest CI build with aod
Turn on sleep mode
Put watch to sleep
Try to wake it up via anything but a button press (whatever u have set it to)
Watch it wake up

### Version

Latest CI build

FintasticMan, 2024-08-23:
Interesting. I'm not able to reproduce this issue with the latest CI build, with AOD on or off. Are you able to determine which commit the issue is introduced in for you?

tituscmd, 2024-08-23:
Sadly I updated my watch only now that the AOD has fully been implemented, so I'm not sure where the issue came from. I also added the PR for improved timer vibration in that update, so the combination might be the issue (or maybe even the PR itself).

mark9064, 2024-11-17:
After the state reworks this should definitely be solved now, for sure re-open/make a new issue if this happens again

closed 2024-11-17 by mark9064