InfiniTimeOrg/InfiniTime #2110: Unresponsive

tituscmd, 2024-08-25:
Latest CI build feels less responsive to me. I usually know where and when to swipe by muscle memory but it feels like I have to wait longer between each swipe since updating.

mark9064, 2024-08-25:
Can you bisect? If not, which build are you comparing to?

tituscmd, 2024-08-25:
I'm comparing to a build of mine from about 2 months ago.
Also the movements that feel less responsive are just any movements.

mark9064, 2024-08-25:
Do you have a commit ref for that build?

tituscmd, 2024-08-25:
Unsure where to find that, but I have the DFU on hand.

mark9064, 2024-08-25:
When the firmware is running you can see the short ref in the settings/about screen

tituscmd, 2024-08-25:
I see, thank you!
The short ref is d6e9361a

mark9064, 2024-08-25:
That ref is to the tip of your casio weather branch (https://github.com/tituscmd/InfiniTime/tree/casio_weather)
I think you might have the wrong ref here, as you said the build is about 2 months old, but the tip of this branch is only 3 days old

tituscmd, 2024-08-25:
Oh, I gave you the wrong one.
This is it: 57e625d4

tituscmd, 2024-08-25:
Is it possible on windows?

mark9064, 2024-08-25:
Yes, I've previously written a DFU Python script that is cross platform, I don't mind sending it here. The setup is manual though. And as long as your PC has a bluetooth adapter of course!

tituscmd, 2024-08-25:
I think tho it's unneccessary because to do any coding with the PineTime I would boot over to linux anyway, so I think I'm good on that behalf! But I have better experience with flashing via InfiniLink, so I think I'm gonna wait for that.

tituscmd, 2024-10-02:
I'm not sure what's different this time, but today I flashed d6e9361a onto my main PineTime and it feels much smoother than my previous version! I don't know what's the difference to last time I reported on this, but I'm certainly not complaining and this issue is as good as solved in my opinion.

Previous version (feels less smooth)
https://github.com/user-attachments/assets/f661472f-8c37-4cc1-b20d-f998b2fa4887

Current version (feels smoother)
https://github.com/user-attachments/assets/854fe479-0047-49d0-bff0-da4eec2a0d9f

closed 2024-10-02 by tituscmd